mavlink/MAVSDK-Python #587: Can't takeoff on Nvidia Jetson Xavier NX

allwili, 2023-04-24:
Hi,
I'm running my control program on a Nvidia Jetson Xavier NX with Ubuntu 20.04 as OS, to control a Pixhawk4 controller for a quadcopter. I'm running this program on python3.8 and the mavsdk server version is 1.4.13.
To connect the program to the drone I execute the command  `sudo ~/.local/lib/python3.8/site-packages/mavsdk/bin/mavsdk_server -p 50051 serial:///dev/ttyACM0` on one terminal, and then run the main program on another terminal.

```
drone = System(port=50051)
await drone.connect()
print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break
print('--Arming')
await drone.action.arm()
print('--Taking off')
```
Just after arming succesfuly,  I execute the instructions to set the target altitude and the takeoff.

```    
await drone.action.set_takeoff_altitude(aTargetAltitude)
await drone.action.takeoff()
await asyncio.sleep(10)
```
or this way,
```
await drone.param.set_param_float('MIS_TAKEOFF_ALT',aTargetAltitude)
await drone.action.takeoff()
await asyncio.sleep(10)
```

But none of this method works to takeoff.
The server terminal gave me this as output for both cases
```
[09:25:04|Warn ] sending again, retries to do: 3  (MIS_TAKEOFF_ALT). (mavlink_parameters.cpp:1340)
[09:25:04|Warn ] sending again, retries to do: 2  (MIS_TAKEOFF_ALT). (mavlink_parameters.cpp:1340)
[09:25:05|Warn ] sending again, retries to do: 1  (MIS_TAKEOFF_ALT). (mavlink_parameters.cpp:1340)
[09:25:05|Error] Error: Retrying failed get param busy timeout: MIS_TAKEOFF_ALT (mavlink_parameters.cpp:1358)
```
And the main program shows me this as error of the first method
```
--Arming
--Taking off
Traceback (most recent call last):
  File "wind_mav_v1.py", line 966, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "wind_mav_v1.py", line 945, in main
    await arm_and_takeoff(2, drone)
  File "wind_mav_v1.py", line 923, in arm_and_takeoff
    await takeoff(aTargetAltitude, drone)
  File "wind_mav_v1.py", line 898, in takeoff
    await drone.action.set_takeoff_altitude(aTargetAltitude)
  File "/home/t2508/.local/lib/python3.8/site-packages/mavsdk/action.py", line 757, in set_takeoff_altitude
    raise ActionError(result, "set_takeoff_altitude()", altitude)
mavsdk.action.ActionError: PARAMETER_ERROR: 'Parameter Error'; origin: set_takeoff_altitude(); params: (2,)
```

and this as the second method

```
--Arming
--Taking off
Traceback (most recent call last):
  File "wind_mav_v1.py", line 966, in <module>
    asyncio.run(main())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "wind_mav_v1.py", line 945, in main
    await arm_and_takeoff(2, drone)
  File "wind_mav_v1.py", line 923, in arm_and_takeoff
    await takeoff(aTargetAltitude, drone)
  File "wind_mav_v1.py", line 897, in takeoff
    await drone.param.set_param_float('MIS_TAKEOFF_ALT',aTargetAltitude)
  File "/home/t2508/.local/lib/python3.8/site-packages/mavsdk/param.py", line 672, in set_param_float
    raise ParamError(result, "set_param_float()", name, value)
mavsdk.param.ParamError: TIMEOUT: 'Timeout'; origin: set_param_float(); params: ('MIS_TAKEOFF_ALT', 2)
```
does any body know how to solve this problem?

julianoes, 2023-04-24:
Are you using PX4 or ArduPilot?

allwili, 2023-04-25:
Ardupilot. 
I think I'm going to change it to PX4