mavlink/MAVSDK-Python #588: Mavsdk is awayting endlessly. How to know which port to connect to?

kramer65, 2023-04-25:
I've been running ardupilot SITL simulations for a while now, and I now want to start doing commands to the simulation from python (because I want to build a companion computer). To do this I created a new virtualenv in which I installed mavsdk. I then started sitl using `sim_vehicle.py --console --map -v ArduCopter` and opened another window in which I started `apython`:
```
>>> from mavsdk import System
>>> drone = System()
>>> await drone.connect()

```
This doesn't do anything. It just sits there doing nothing. I then found that the address and port can be different, so I checked which ports are used using netstat:
```
$ sudo netstat -tulpn | grep ardu
tcp        0      0 0.0.0.0:5760            0.0.0.0:*               LISTEN      114722/arducopter   
tcp        0      0 0.0.0.0:5762            0.0.0.0:*               LISTEN      114722/arducopter   
tcp        0      0 0.0.0.0:5763            0.0.0.0:*               LISTEN      114722/arducopter   
udp        0      0 0.0.0.0:5501            0.0.0.0:*                           114722/arducopter
```
I tried all the ports above (`5760`,  `5762`, `5763` and `5501`) plus some others using the following code:
```
>>> from mavsdk import System
>>> drone = System(mavsdk_server_address='127.0.0.1', port=5501)
>>> await drone.connect()
```

But it always just halts/freezes there with no output. I'm on Ubuntu 22.04 and python 3.10 (if that matters).

Does anybody know what I'm doing wrong here? Any tips on how to debug this are welcome!

JonasVautherin, 2023-04-25:
Can you try the following?

```python
from mavsdk import System
drone = System()
await drone.connect(system_address="tcp://127.0.0.1:5760")
```

`mavsdk_server_address` in `System()` is for when you start `mavsdk_server` manually. `system_address` in `connect()` is to tell the embedded `mavsdk_server` how to connect to the drone.

I don't use Ardupilot but I think that it should be 5760.

kramer65, 2023-04-26:
Awesome, that was it. Thanks @JonasVautherin! 

For future readers (including myself); it turned out to be port 5762. 

One more question for my understanding. You're saying the extra parameters in `System()` which I used where for when you start the `mavsdk_server` manually. Since I don't do that, does that mean that this actually starts the `mavsdk_server` automatically in the background, which is why we need to use `apython` instead of the regular python repl?

JonasVautherin, 2023-04-26:
Almost :wink:.

You are right that when you install MAVSDK with `pip install mavsdk`, you get an embedded `mavsdk_server` binary which is started automatically in the background when you use `System()`.

However, `apython` is unrelated. It is required because MAVSDK-Python using `asyncio` (from the Python standard library), and the regular Python REPL cannot run `asyncio` code directly. `apython` is just some kind of wrapper that makes it easier to use `asyncio` in the REPL.

Does that make sense?

kramer65, 2023-05-02:
Thanks for clearing that up. Good to know that I don't need to run the server separately.

In the end I ran into another error when trying to take off:
```
>>> from mavsdk import System
>>> drone = System()
>>> await drone.connect(system_address="tcp://127.0.0.1:5762")
>>> await drone.action.arm()
>>> await drone.action.takeoff()
Traceback (most recent call last):
  File "/home/kramer65/Downloads/mavsdk-test/venv/lib/python3.10/site-packages/aioconsole/execute.py", line 145, in aexec
    result, new_local = await coro
  File "<console>", line 2, in __corofn
  File "/home/kramer65/Downloads/mavsdk-test/venv/lib/python3.10/site-packages/mavsdk/action.py", line 368, in takeoff
    raise ActionError(result, "takeoff()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: takeoff(); params: ()
```
Any idea what's wrong here?

JonasVautherin, 2023-05-02:
The autopilot returned `COMMAND_DENIED`, for some reason (maybe it's not ready, maybe it is waiting for a GPS fix, etc). That's not a MAVSDK issue, it's really just MAVSDK reporting the error that was sent by the autopilot :+1:.

kramer65, 2023-05-04:
Alright thanks. Then I'll proceed digging for the problem. Thanks for all the help!

closed 2023-05-04 by kramer65