mavlink/MAVSDK-Python #590: set position,velocity and accelaration offboard setpoint

alireza787b, 2023-05-16:
Hi. As far as I see I can either send acceleration offboard or position, velocity offboard is that correct? Can we just send all? I can see that px4 supports it and we can achieve similar behavior using Mavros.
I want to send offboard setpoints for position and feedforward the acceleration and velocity as well.
if this function is not available, how can I implement it myself? I think we should chang ethe mavlink command that is being sent but since everything is auto-generated I don't know how and what to look for

julianoes, 2023-05-17:
Yes, that's correct. The closest API for you would be position with velocity feedforward. If that's not enough, consider adding it.

First step is to add the API that you would like here:
https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/offboard/offboard.proto

alireza787b, 2023-05-17:
Is there documentation or explanation and example of how to add this? I don't know he concepts behind that which parts I should edit and which part is auto-generated.

julianoes, 2023-05-18:
The first step is the proto file, then auto-generation happens from that in the next step.

Also see: https://mavsdk.mavlink.io/main/en/cpp/contributing/plugins.html

alireza787b, 2023-05-18:
Thanks I already read that doc and almost did all it said. I created my custom function and it should work. this is the function in cpp:
```
Offboard::Result OffboardImpl::set_position_velocity_acceleration_ned(
    Offboard::PositionNedYaw position_ned_yaw, Offboard::VelocityNedYaw velocity_ned_yaw, Offboard::AccelerationNed acceleration_ned)
{
    {
        std::lock_guard<std::mutex> lock(_mutex);
        _position_ned_yaw = position_ned_yaw;
        _velocity_ned_yaw = velocity_ned_yaw;
        _acceleration_ned = acceleration_ned;

        if (_mode != Mode::PositionVelocityAccelerationNed) {
            if (_call_every_cookie) {
                // If we're already sending other setpoints, stop that now.
                _system_impl->remove_call_every(_call_every_cookie);
                _call_every_cookie = nullptr;
            }
            // We automatically send Ned setpoints from now on.
            _system_impl->add_call_every(
                [this]() { send_position_velocity_acceleration_ned(); }, SEND_INTERVAL_S, &_call_every_cookie);

            _mode = Mode::PositionVelocityAccelerationNed;
        } else {
            // We're already sending these kind of setpoints. Since the setpoint change, let's
            // reschedule the next call, so we don't send setpoints too often.
            _system_impl->reset_call_every(_call_every_cookie);
        }
    }

    // also send it right now to reduce latency
    return send_position_velocity_acceleration_ned();
}

```

and this is in .h
```
 mutable std::mutex _mutex{};
    enum class Mode {
        NotActive,
        PositionNed,
        // PositionGlobalAMSL,
        PositionGlobalAltRel,
        VelocityNed,
        PositionVelocityNed,
        PositionVelocityAccelerationNed,
        AccelerationNed,
        VelocityBody,
        Attitude,
        AttitudeRate,
        ActuatorControl
    } _mode = Mode::NotActive;
```

`    Offboard::Result send_position_velocity_acceleration_ned();
`

```
 Offboard::Result set_position_velocity_acceleration_ned(
        Offboard::PositionNedYaw position_ned_yaw, Offboard::VelocityNedYaw velocity_ned_yaw, Offboard::AccelerationNed acceleration_ned);
```

but when I try to build it in the last step in 100% I get the following error:

[100%] Building CXX object src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/offboard_service_impl_test.cpp.o
In file included from /home/alireza/MAVSDK/src/mavsdk_server/test/offboard_service_impl_test.cpp:5:
/home/alireza/MAVSDK/src/mavsdk_server/test/../src/plugins/offboard/offboard_service_impl.h: In instantiation of ‘grpc::Status mavsdk::mavsdk_server::OffboardServiceImpl<Offboard, LazyPlugin>::SetPositionVelocityAccelerationNed(grpc::ServerContext*, const mavsdk::rpc::offboard::SetPositionVelocityAccelerationNedRequest*, mavsdk::rpc::offboard::SetPositionVelocityAccelerationNedResponse*) [with Offboard = testing::NiceMock<mavsdk::testing::MockOffboard>; LazyPlugin = testing::NiceMock<mavsdk::mavsdk_server::testing::MockLazyPlugin<testing::NiceMock<mavsdk::testing::MockOffboard> > >]’:
/home/alireza/MAVSDK/src/mavsdk_server/test/../src/plugins/offboard/offboard_service_impl.h:959:18:   required from here
/home/alireza/MAVSDK/src/mavsdk_server/test/../src/plugins/offboard/offboard_service_impl.h:984:52: error: ‘class testing::NiceMock<mavsdk::testing::MockOffboard>’ has no member named ‘set_position_velocity_acceleration_ned’; did you mean ‘set_position_velocity_ned’?
  984 |         auto result = _lazy_plugin.maybe_plugin()->set_position_velocity_acceleration_ned(translateFromRpcPositionNedYaw(request->position_ned_yaw()), translateFromRpcVelocityNedYaw(request->velocity_ned_yaw()), translateFromRpcAccelerationNed(request->acceleration_ned()));
      |                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                       set_position_velocity_ned
cc1plus: note: unrecognized command-line option ‘-Wno-unknown-warning-option’ may have been intended to silence earlier diagnostics
gmake[2]: *** [src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/build.make:160: src/mavsdk_server/test/CMakeFiles/unit_tests_mavsdk_server.dir/offboard_service_impl_test.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:2175: src

Should I do extra step in mavsdk_server files?
I think I can hard code the position_velocity to include ACC as well and it should work but I am planing to create a new function and then pull it to repo so others can use it later

julianoes, 2023-05-18:
Can you make the PR for the proto changes, so that I can review those first?

For the mock errors, make sure to add the API here:
https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/offboard/mocks/offboard_mock.h

alireza787b, 2023-05-18:
editing the mock file was the step missing in the docs and it solved it. it is not generated in mavsdk cpp. now I want to generate it for python as well. this is I don't know how. 
My forked mavsdk repo is at :
https://github.com/alireza787b/MAVSDK

Would you like me to send PR on mavsdk or mavsdk python?
I need help with this converting to Python part. I haven't found any documentation on that.

JonasVautherin, 2023-05-18:
> Would you like me to send PR on mavsdk or mavsdk python?

First one would be `MAVSDK-Proto`. That will generate the C++ headers, which you need to implement. Python is then fully generated.

alireza787b, 2023-05-19:
I managed to genreate python and it shows up but when I run the new function in python I get multithreading or sth like that .
I created a pull request on mavsdk proto and describe the steps I did and the problem
this is the link:
https://github.com/mavlink/MAVSDK-Proto/pull/314

PS:
sth is still unclear for me. I edited the cpp sources in mavsdk repo. how they end up effecting the python? how the system exactly works? I changed the proto, build the mavsdk and then did python generation from mavsdk python.

Thanks for your help

naturalspirit299, 2023-07-15:
Hello! You are correct that in offboard mode, you can send either acceleration or position/velocity setpoints. However, it is indeed possible to send all three types of setpoints simultaneously if desired. PX4 and Mavros support this capability, allowing you to achieve the desired behavior.

If the function for sending offboard setpoints with position, velocity, and acceleration is not readily available, you can implement it yourself. One way to do this is by modifying the Mavlink command that is being sent. Since the code is auto-generated, you might need to search for the specific location where the setpoint commands are handled.

Start by locating the part of the code where Mavlink messages are generated and sent to the flight controller. Look for functions or sections related to offboard setpoints, such as `set_position_target_local_ned` or `set_position_target_global_int`. These functions typically handle position and velocity setpoints.

To include acceleration in your setpoint message, you would need to modify the existing Mavlink message structure and include the appropriate acceleration fields. You may need to refer to the Mavlink message definitions to ensure compatibility.

Once you have made the necessary changes to include acceleration in the setpoint message, rebuild or recompile the code, and deploy it to your system. Make sure to test thoroughly to ensure the desired behavior is achieved.

Remember to consult the PX4 and Mavros documentation and community forums for further guidance and support during the implementation process. Good luck!

alireza787b, 2023-07-16:
> Hello! You are correct that in offboard mode, you can send either acceleration or position/velocity setpoints. However, it is indeed possible to send all three types of setpoints simultaneously if [...]

Thanks for the information and guide
I did that and added a new function for setting all position, velocity, and acceleration with command and it works ok 
the link to the pull request is :
https://github.com/mavlink/MAVSDK/pull/2057

and in it available in the latest mavsdk release