mavlink/MAVSDK-Python #591: action.do_orbit no attribut found error message.

arpitpara, 2023-05-17:
Hi,
I am trying to achieve an auto mission with do_orbit function to move the drone in circle facing towards alawys front option.
I tried using mavsdk python and it gives the following error.

`Error msg:
Waiting for drone to connect...
Drone discovered!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
--- Taking Off
Traceback (most recent call last):
  File "/home/lalith/MAVSDK-Python/testScripts/windmillinspect.py", line 43, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
[image]

  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "/home/lalith/MAVSDK-Python/testScripts/windmillinspect.py", line 30, in run
    await drone.action.do_orbit(radius_m = 5,
  File "/home/lalith/MAVSDK-Python/drones/lib/python3.8/site-packages/mavsdk/action.py", line 587, in do_orbit
    request.yaw_behavior = yaw_behavior.translate_to_rpc()
AttributeError: 'int' object has no attribute 'translate_to_rpc'`

JonasVautherin, 2023-05-17:
Looks like it could be a bug indeed. Did you check the code at /home/lalith/MAVSDK-Python/drones/lib/python3.8/site-packages/mavsdk/action.py", line 587? Can you show it here?

arpitpara, 2023-05-18:
`import asyncio
from mavsdk import System
import mavsdk
from mavsdk.action import OrbitYawBehavior

async def run():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("Drone discovered!")
            break

    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break
    
    print("-- Arming")
    await drone.action.arm()

    print("--- Taking Off")
    await drone.action.takeoff()
    await asyncio.sleep(5)

    await drone.action.do_orbit(radius_m = 5,
                                velocity_ms = 1,
                                yaw_behavior = OrbitYawBehavior.HOLD_FRONT_TO_CIRCLE_CENTER,                                
                                latitude_deg = 47.398039859999997,
                                longitude_deg = 8.5455725400000002,
                                absolute_altitude_m = 15)
    await asyncio.sleep(20)

    await drone.action.land()
    print("--- Landing")

if __name__ == "__main__":
    # asyncio.run(run())

    loop = asyncio.get_event_loop()
    loop.run_until_complete(run())`

I am using like this.. This also didn't worked.

julianoes, 2023-05-18:
Which PX4 version is this against?

arpitpara, 2023-05-18:
My action.py is
`# -*- coding: utf-8 -*-
# DO NOT EDIT! This file is auto-generated from
# https://github.com/mavlink/MAVSDK-Python/tree/main/other/templates/py
from ._base import AsyncBase
from . import action_pb2, action_pb2_grpc
from enum import Enum

class OrbitYawBehavior(Enum):
    """
     Yaw behaviour during orbit flight.

     Values
     ------
     HOLD_FRONT_TO_CIRCLE_CENTER
          Vehicle front points to the center (default)

     HOLD_INITIAL_HEADING
          Vehicle front holds heading when message received

     UNCONTROLLED
          Yaw uncontrolled

     HOLD_FRONT_TANGENT_TO_CIRCLE
          Vehicle front follows flight path (tangential to circle)

     RC_CONTROLLED
          Yaw controlled by RC input

     """

    
    HOLD_FRONT_TO_CIRCLE_CENTER = 0
    HOLD_INITIAL_HEADING = 1
    UNCONTROLLED = 2
    HOLD_FRONT_TANGENT_TO_CIRCLE = 3
    RC_CONTROLLED = 4

    def translate_to_rpc(self):
        if self == OrbitYawBehavior.HOLD_FRONT_TO_CIRCLE_CENTER:
            return action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_FRONT_TO_CIRCLE_CENTER
        if self == OrbitYawBehavior.HOLD_INITIAL_HEADING:
            return action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_INITIAL_HEADING
        if self == OrbitYawBehavior.UNCONTROLLED:
            return action_pb2.ORBIT_YAW_BEHAVIOR_UNCONTROLLED
        if self == OrbitYawBehavior.HOLD_FRONT_TANGENT_TO_CIRCLE:
            return action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_FRONT_TANGENT_TO_CIRCLE
        if self == OrbitYawBehavior.RC_CONTROLLED:
            return action_pb2.ORBIT_YAW_BEHAVIOR_RC_CONTROLLED

    @staticmethod
    def translate_from_rpc(rpc_enum_value):
        """ Parses a gRPC response """
        if rpc_enum_value == action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_FRONT_TO_CIRCLE_CENTER:
            return OrbitYawBehavior.HOLD_FRONT_TO_CIRCLE_CENTER
        if rpc_enum_value == action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_INITIAL_HEADING:
            return OrbitYawBehavior.HOLD_INITIAL_HEADING
        if rpc_enum_value == action_pb2.ORBIT_YAW_BEHAVIOR_UNCONTROLLED:
            return OrbitYawBehavior.UNCONTROLLED
        if rpc_enum_value == action_pb2.ORBIT_YAW_BEHAVIOR_HOLD_FRONT_TANGENT_TO_CIRCLE:
            return OrbitYawBehavior.HOLD_FRONT_TANGENT_TO_CIRCLE
        if rpc_enum_value == action_pb2.ORBIT_YAW_BEHAVIOR_RC_CONTROLLED:
            return OrbitYawBehavior.RC_CONTROLLED

    def __str__(self):
        return self.name

class ActionResult:
    """
     Result type.
[... 796 lines cut ...]

         This will set the speed during a mission, reposition, and similar.
         It is ephemeral, so not stored on the drone and does not survive a reboot.

         Parameters
         ----------
         speed_m_s : float
              Speed in meters/second

         Raises
         ------
         ActionError
             If the request fails. The error contains the reason for the failure.
        """

        request = action_pb2.SetCurrentSpeedRequest()
        request.speed_m_s = speed_m_s
        response = await self._stub.SetCurrentSpeed(request)

        
        result = self._extract_result(response)

        if result.result != ActionResult.Result.SUCCESS:
            raise ActionError(result, "set_current_speed()", speed_m_s)
        `

arpitpara, 2023-05-18:
> Which PX4 version is this against?

I am using px4 sitl with gazebo simulation. Try checking the version in QGC.
In heartbeat it only show autopilot = 12.
No autopilot version is visible.

dayjaby, 2023-05-18:
If you do `git log` in the PX4 folder, you can find out which version your PX4 is based on.

Can you be more precise what exactly is not working? Usually it's good to post the entire error messages (and not entire code files) here to help find the issue. At least your initial issue should be solved:

```
>>> OrbitYawBehavior.HOLD_FRONT_TO_CIRCLE_CENTER.translate_to_rpc()
0
```

arpitpara, 2023-05-18:
> If you do `git log` in the PX4 folder, you can find out which version your PX4 is based on.
>
> Can you be more precise what exactly is not working? Usually it's good to post the entire error messages [...]

Sorry for pasting the whole code. But there was no error shown.  That's why

julianoes, 2023-05-18:
No error but it doesn't work? Does the drone go into orbit mode when you observe it with QGC?

arpitpara, 2023-05-18:
> No error but it doesn't work? Does the drone go into orbit mode when you observe it with QGC?

Yes It doesn't work.
It Just takes off and land as soon as it is off the ground.
And Yes It does change to orbit mode while observing in QGC.

dayjaby, 2023-05-18:
But it doesn't perform the orbit flight despite being in orbit mode. Is that "what's not working"? In that case you should create the issue in PX4 itself: https://github.com/PX4/PX4-Autopilot and make sure to attach the log file of the simulation flight, and of course your MAVSDK script. The log should be in ~/.ros/sitl_<your airframe type>_0/log

arpitpara, 2023-05-19:
> But it doesn't perform the orbit flight despite being in orbit mode. Is that "what's not working"? In that case you should create the issue in PX4 itself: https://github.com/PX4/PX4-Autopilot and make sure to attach the log file of the simulation flight, and of course your MAVSDK script. The log should be in ~/.ros/sitl__0/log

ok Thank you!.
Are you sure it's not a bug of mavsdk-python?

arpitpara, 2023-05-19:
> ok Thank you!.
And Yes it's not performing orbit operation that we're giving.
 Are you sure it's not a bug of mavsdk-python?

dayjaby, 2023-05-19:
If the drone gets into orbit mode, MAVSDK did its job: Forward the command correctly to PX4. Or was the drone landing too quickly, so not waiting 20 seconds before going into land mode?

dayjaby, 2023-05-19:
So can you share the .ulg file of the simulation flight (as mentioned in ~/.ros/sitl_..../logs` folder). That should help a lot with debugging this!

arpitpara, 2023-05-19:
I got this through QGC!
[log_0_2023-5-19-17-30-46.zip](https://github.com/mavlink/MAVSDK-Python/files/11516642/log_0_2023-5-19-17-30-46.zip)

dayjaby, 2023-05-19:
[image]

so....analyzing this ulg file with plotjuggler: The drone was for ~10 seconds in ORBIT mode. Are you sure that the code is the same, because in your code you wait 20 seconds after commanding the orbit mode. The navigation mode is written in the `vehicle_status` uORB topic in the field `nav_state`. As you can see, it's 21 for around 10 seconds and checking https://github.com/PX4/PX4-Autopilot/blob/main/msg/VehicleStatus.msg you will find `NAVIGATION_STATE_ORBIT = 21`. 

The issue seems that the takeoff didn't have enough time to do it's job: the drone reached only 0.5m altitude. Then you gave him the orbit command to orbit somewhere at 2.5m AMSL, even though your flight started at ~488m AMSL.

So my tip: Use `float("nan")` for param7/altitude to use the current vehicle altitude (see https://mavlink.io/en/messages/common.html#MAV_CMD_DO_ORBIT) and give all the sleeps a little bit more time, especially the takeoff. Or consider a way of checking whether the drone finished takeoff before sending the orbit command.

julianoes, 2023-05-20:
@arpitpara it would be great if you could add an example to https://github.com/mavlink/MAVSDK-Python/tree/main/examples once you have something that nicely works against SITL.