mavlink/MAVSDK-Python #592: Unable to connect with drone while using the UDP of "Remote IP Address".

MuhammadBilal1, 2023-05-17:
Hi,
    I am trying to connect with my drone using the "remote IP address" of UDP.
When I connect the PX4 simulator, it works fine.
    ```await drone.connect(system_address="udp://14540")```

### I tried the following patterns, but none of them worked. 
    await drone.connect(system_address="udp://10.17.111.1:40000")
    await drone.connect(system_address="udp:10.17.111.1:40000")
    await drone.connect(system_address="udpout://10.17.111.1:40000//")
    await drone.connect(system_address="udp://10.17.111.1")

I am using the remote host (10.17.111.1), with the following details:
**IP:** 10.17.111.1
**Port:** 40000

I am connecting it while using the VPN. 

### **Mavproxy works fine, I am getting heartbeat:**
[image]

### **Working fine with QGroundControl, I am getting everything on QGC:**
[image]

JonasVautherin, 2023-05-17:
This one should work: `await drone.connect(system_address="udp://10.17.111.1:40000")`

Which version of MAVSDK are you using?

MuhammadBilal1, 2023-05-17:
> This one should work: `await drone.connect(system_address="udp://10.17.111.1:40000")`
> 
> Which version of MAVSDK are you using?

I am using **MAVSDK==1.4.4**.

JonasVautherin, 2023-05-17:
Hmm I don't remember clearly, but it could be that this doesn't work with MAVSDK v1. It works with the `main` branch though... but you have to compile `mavsdk_server` from source and use it with MAVSDK-Python from source, too.

MuhammadBilal1, 2023-05-21:
> Hmm I don't remember clearly, but it could be that this doesn't work with MAVSDK v1. I am pretty sure that it works with the `main` branch though (which will eventually be released as MAVSDK v2)... but you have to compile `mavsdk_server` from source and use it with MAVSDK-Python from source, too.

I am trying to **compile** mavsdk_server from the **main** branch using the source code by following the instructions given at [Build mavsdk_server from Source](https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html). But I am getting errors while building.

Environment: Ubuntu 20.04

Here is the sequence of commands I am running to build mavsdk_server from source:
**Method # 1:** ([Build from source](https://mavsdk.mavlink.io/main/en/cpp/guide/build_mavsdk_server.html))
```
sudo apt-get update
sudo apt-get install build-essential cmake git
git clone https://github.com/mavlink/MAVSDK.git
git submodule update --init --recursive
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_SHARED_LIBS=OFF -DBUILD_MAVSDK_SERVER=ON -Bbuild/default -H.
cmake --build build/default -j8
```
**Below is the output of the last two commands:**
[image]
[image]

**Method # 2** ([Release Build & Local Install](https://mavsdk.mavlink.io/main/en/cpp/guide/build.html#requirements))
```
sudo apt-get update
sudo apt-get install build-essential cmake git
git clone https://github.com/mavlink/MAVSDK.git
git submodule update --init --recursive
cmake -Bbuild/default -DCMAKE_BUILD_TYPE=Release -H.
cmake --build build/default -j8
cmake -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=/home/bilal/mavsdk/installation/ -Bbuild/default -H.
cmake --build build/default --target install
```

After following the instruction, I don't know what should I do next to start mavsdk_server!
I tried to search for the _mavsdk_server_, but i was unable to find that.

Can you please guide me what should I do and which method should I follow to proceed?

JonasVautherin, 2023-05-21:
Your log should find openssl in `third_party/` (like the other dependencies), but it found it on the system (I cannot copy-paste the line because it is a screenshot :innocent:).

You need to understand why that it. Probably an error when it built OpenSSL, which should appear earlier in your log.

MuhammadBilal1, 2023-05-21:
> 

So the **Method # 1** is the correct approach?
I will check the openssl issue

julianoes, 2023-05-21:
https://github.com/mavlink/MAVSDK-Python/issues/592#issuecomment-1551609565

This works with v1.4.4. I don't think you need to build the server yourself.

MuhammadBilal1, 2023-05-22:
@julianoes @JonasVautherin 
Can you please guide me on how I can move forward? This is very critical for us.

In the below screenshot, I am showing you the following:

- mavproxy is connecting to the same address successfully.
- I can ping the UDP port using netcat.
- mavsdk installed version is 1.4.4 (although I tried on 1.4.6, but output is same)
- I am trying to connect with the drone at my udp address, but mavsdk is waiting forever.

[image]

JonasVautherin, 2023-05-22:
You have a typo in your code. It should read `udp://10.17.111.1:40000` (your code is missing `//`).

Then are you certain that both mavproxy and MAVSDK can connect at the same time? I have seen plenty of mavlink implementations where the UDP server (i.e. the program listening on port 40000 here) only ever answers to the first client (here it would be mavproxy, I suppose) and ignores the others. I have also seen implementations where the server answers to the last client (so it would jump between mavproxy and MAVSDK, each receiving a subset of the messages. But that is apparently not happening here, because MAVSDK would detect the drone in that case).

MuhammadBilal1, 2023-05-22:
Sorry, I was trying to test different combinations and sent the screenshot of different combination (where I removed the ```//```). Sending you the updated screenshot as well.
I also had experience with the scenario that multiple connections will not work as you said. Like MAVSDK & QGroundControl can't work together at the same time.

I actually, stopped other things like QGroundControl, mavproxy to test this combination.
None of the other task is running. Below screenshot shows only the MAVSDK is running.

[image]

JonasVautherin, 2023-05-22:
Can you check with wireshark if heartbeats are being sent by MAVSDK to 10.17.111.1:40000?

JonasVautherin, 2023-05-22:
Seems like MAVSDK is sending the heartbeats, then... :thinking:

julianoes, 2023-05-22:
PX4 doesn't switch the remote port once it has received UDP from one place.

You could try if this change helps at all: https://github.com/PX4/PX4-Autopilot/pull/21608