mavlink/MAVSDK-Python #596: Drone Id

arpitpara, 2023-05-23:
Hi
How do I find the drone id through mavsdk code.?
Whas is the process of generating a drone id.?

JonasVautherin, 2023-05-23:
How does that work with MAVLink? Is there a MAVLink message giving the drone id?

arpitpara, 2023-05-23:
yes. There are messages that gives information about the remote id of the drone and related information.
Please check.[https://mavlink.io/en/services/opendroneid.html](url)

dayjaby, 2023-05-23:
If you want to differentiate between multiple drones, you can use different "system id"s. You can check https://mavlink.io/en/guide/message_signing.html how this will change the mavlink messages on byte-level.

To do so in PX4, change the MAV_SYS_ID parameter to different values. In MAVSDK you can do different stuff when connecting a new system based on the system id:

```
    mavsdk.subscribe_on_new_system([&]() {
        for (auto& system : mavsdk.systems())
        {
            uint8_t system_id = system->get_system_id();
            if (system_id >= 250) {
                found_gcs = true;
            } else if (system->has_autopilot()) {
                found_autopilot = true;
            }
            if (found_gcs && found_autopilot) {
                mavsdk.subscribe_on_new_system(nullptr); // stop subscribing to new systems as we got all systems that we want
            }
        }
    }):
    ... use mavsdk.systems() here ...
```

closed 2023-05-25 by julianoes