mavlink/MAVSDK-Python #597: How is the TrackingServer API used?

Bharannn, 2023-05-25:
Hi,
How is the TrackingServer API used? as there is no example script available for it, I'm guessing if i give the bounding box top-left and bottom-right coordinates of a detected object, the drone can track the object. I don't know if I'm right or not, but please let me know how I can work on it, or provide any example script.

JonasVautherin, 2023-05-25:
> I'm guessing if i give the bounding box top-left and bottom-right coordinates of a detected object, the drone can track the object.

I don't think so :see_no_evil:. It is a MAVSDK "server", meaning that it will help you implement the drone side of it. But MAVSDK cannot automagically know about your specific camera and autopilot and make it just work.

Interestingly, MAVSDK doesn't have a tracking plugin (unless I missed something), so you can't talk to that server from MAVSDK. So if you want to use it as a client, you probably need to contribute a tracking (client) plugin :innocent:.

If you look at the [server implementation](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/tracking_server/tracking_server_impl.cpp#L26-L45), it's apparently "just" about supporting those message:

* MAV_CMD_CAMERA_TRACK_POINT
* MAV_CMD_CAMERA_TRACK_RECTANGLE
* MAV_CMD_CAMERA_STOP_TRACKING

Hope it helps! :blush:

Bharannn, 2023-05-25:
So if I deploy this script on the drone giving it inout of bounding box of recatangles detected through python code from my camera. Will it work?

JonasVautherin, 2023-05-25:
Hmm, I think that the point of the [MAV_CMD_CAMERA_TRACK_RECTANGLE](https://mavlink.io/en/messages/common.html#MAV_CMD_CAMERA_TRACK_RECTANGLE) is that the GCS sends it to the camera, and then the camera tracks it.

So in your code implementing a tracking server, you should receive the rectangle command, and forward it to your camera with whatever API your camera has.

Does that make sense?

Bharannn, 2023-05-25:
Umm I guess yes. You're saying tracking server doesn't relate anything to giving commands to drone movement to track the rectangle.
Example - if a person got detected tracking server doesn't tell drone to follow the person on the basis of rectangle input.
It only tells camera to follow the person in the frames.

it doesn't make sense in this case. 
What I was thinking is I will be able to feed rectangle input for tracking command and the function will decide to navigate the drone automatically to track it.

JonasVautherin, 2023-05-25:
A MAVSDK server is just some kind of "framework" to help you write your server. It will just receive the MAVLink message, tell you "I received this message", and let you do whatever you want with it. It gives you helpers to answer with MAVLink messages (in case it's a command, or something).

You are the one who needs to do something with this MAVLink message. MAVSDK cannot automagically detect the components running next to itself and start sending them instructions to "follow a person" :thinking:. You have to write code that will do that :grin:.

closed 2023-08-07 by Bharannn