mavlink/MAVSDK-Python #606: About gimbal position

reliablebull, 2023-06-20:
Hi, I have a few questions.

Gimbal pitch and yaw are printed twice on gimbal.py . Is it working properly?

Thank you.
[image]

JonasVautherin, 2023-06-20:
Hmm it should be printing once per mavlink message :thinking:. What drone is that? Did you check the MAVLink messages coming from the drone?

reliablebull, 2023-06-20:
Thank you for your quick reply.

I'm using px4 sitl typoon_h480.

Thank you.

julianoes, 2023-06-21:
I tested this with latest MAVSDK-Python against PX4 main and it looks correct to me:

```
./gimbal.py 
Taking control of gimbal
Gimbal pitch: 5.776486396789551, yaw: 0.004995898809283972
Setting gimbal mode
Look forward first
Gimbal pitch: 5.77648401260376, yaw: -0.0004088433925062418
Gimbal pitch: 5.77648401260376, yaw: -0.003173567820340395
Gimbal pitch: 5.776483058929443, yaw: 0.000808312906883657
Gimbal pitch: 5.7764811515808105, yaw: 0.0006025618058629334
Gimbal pitch: 5.776482105255127, yaw: -0.003063811454921961
Gimbal pitch: 5.7764787673950195, yaw: 0.0022326591424643993
...
```

What PX4 version are you using?

reliablebull, 2023-06-21:
Thank you for your quick reply.

After reinstalling PX4 firmware, it works normally.

I have a few questions.

Firstly, is the initial value pitch: 5.xx degrees and yaw: 0 degrees correct?
[image]

Secondly, if the pitch is set to -90 degrees, there is noise in the yaw value.
[image]

Thank you.

julianoes, 2023-06-21:
-90 degrees is vertically down where you get a [gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock) and hence noise in the yaw angle if you don't implement the right euler angle conversion for that case.

Sounds like it's all good then.

closed 2023-06-21 by julianoes