mavlink/MAVSDK-Python #608: trying to run quick start for MAVSDK-Python

WCjai, 2023-06-21:
hello i installed the MAVSDK-python as per installation guide, after that i try to do quick start from the MAVSDK guide website (https://mavsdk.mavlink.io/main/en/python/quickstart.html), when i ran "await drone.connect()" it shows error, i tried to search other post simular, but couldn't find solution 

[image]

JonasVautherin, 2023-06-21:
How did you install? With `pip`? Which version is that?

Also what kind of computer is it? I mean the CPU architecture?

WCjai, 2023-07-19:
tried both way of installing, using pip and also build from source, iam using a laptop running ubuntu 20.04 , CPU Architecture:                    x86_64

arpitpara, 2023-07-26:
are you using mavsdk and mavsdk-python together?
Try making a virtual environment and then install evrything fresh related to mavsdk-python.

JonasVautherin, 2023-07-26:
Can you try to download and run `mavsdk_server_musl_x86_64` from https://github.com/mavlink/MAVSDK/releases/tag/v1.4.16? What does it say?

arpitpara, 2023-07-26:
I checked it. And it works fine.
[image]
[image]
[image]

JonasVautherin, 2023-07-26:
@arpitpara: do you also have the error when using `pip`? i.e. `OSError: [Errno 8] Exec format error: '/home/wcjai/MAVSDK-Python/mavsdk/bin/mavsdk_server'`?

arpitpara, 2023-07-27:
No.Everything seems to work fine.

JonasVautherin, 2023-07-27:
Right :rofl:. Well then I was asking @WCjai to try the musl executable :grin:

arpitpara, 2023-07-27:
> Right rofl. Well then I was asking @WCjai to try the musl executable 😁

Better you quote right person next time. I was also in little dilema till now.

WCjai, 2023-07-27:
Sorry guys, I am currently out of town but will try the solutions and provide the information here

WCjai, 2023-08-02:
> Can you try to download and run `mavsdk_server_musl_x86_64` from https://github.com/mavlink/MAVSDK/releases/tag/v1.4.16? What does it say?

thank you all, the server works fine after creating new python environment the program is now connecting to the drone

closed 2023-08-02 by WCjai