mavlink/MAVSDK-Python #610: failed to switch from takeoff to offboard_velocity_ned_yaw

rishireddi, 2023-06-22:
so I'm trying to use the offboard_velocity_ned to control the final landing of my drone to the desired position. But it just lands without any movements to neither north or south nor east or west, after I switch to offboard mode using the example here https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_velocity_ned.py 
Could you please help me with this.
Thank you in advance.

julianoes, 2023-06-22:
Can you share a logfile where you were testing this so that I can see what is going on? Thanks.

rishireddi, 2023-06-23:
Hi @julianoes, Thanks for the response. I am unable to share .ulg file, could you please suggest me any other way if possible.
Thank you.

julianoes, 2023-06-24:
> I am unable to share .ulg

Why?

rishireddi, 2023-06-27:
Hi @julianoes, I am sorry for the delay and I am sharing them now.
[offboard_velocity_ned.zip](https://github.com/mavlink/MAVSDK-Python/files/11878901/offboard_velocity_ned.zip)
Thank you.

julianoes, 2023-06-27:
Please just upload to logs.px4.io and share the link here.

rishireddi, 2023-06-28:
Hi @julianoes I've uploaded them to the link you have told me and this is the link of the flight review.

rishireddi, 2023-06-28:
Hi @julianoes I've uploaded ulog files to the logs.px4.io and the below is the link of the flight review.

(https://logs.px4.io/plot_app?log=d847787a-d900-48d7-8f7c-7815aee067fa)

Thank you.

julianoes, 2023-06-28:
I'm not sure what I'm looking at. The drone has not taken off and is always in Loiter mode.

rishireddi, 2023-06-29:
This issue got solved by changing a few parameters related to offboard mode in the QGC.
@julianoes , thank you for the help.

julianoes, 2023-06-30:
Great. Feel free to share which params needed changing for any future readers. Thanks.

closed 2023-06-30 by julianoes