mavlink/MAVSDK-Python #612: Trying to use ftp, get protocol error

itayg-spearuav, 2023-07-09:
Hi, 
I am trying to use mavsdk's Ftp functionality to list the contents of a directory, among other things such as upload files, create dirs etc. However, I am getting the following error when trying to run my code:
```
File "/home/itayg/Documents/code/mav_ftp_test/mavftp_send.py", line 29, in list_dir
    dirs = await self.mav.ftp.list_directory(path)
  File "/home/itayg/Documents/code/mav_ftp_test/venv/lib/python3.10/site-packages/mavsdk/ftp.py", line 465, in list_directory
    raise FtpError(result, "list_directory()", remote_dir)
mavsdk.ftp.FtpError: PROTOCOL_ERROR: 'Protocol Error'; origin: list_directory(); params: ('/',)
```

My code:

```#!/usr/bin/env python3
from mavsdk import System
import asyncio

class MavFtpHandler:
    mav = None
    address = "serial:///dev/ttyACM0"

    async def init_connection(self):
        self.mav = System(mavsdk_server_address="0.0.0.0")
        print("Connecting to px4...")
        # Connect
        await self.mav.connect(system_address=self.address)
        # This waits till a mavlink based drone is connected
        async for state in self.mav.core.connection_state():
            if state.is_connected:
                print(".. Connected to px4!")
                break
        print()

    async def upload(self, path, remote):
        progress = self.mav.ftp.upload(path, remote)
        async for p in progress:
            print(".")
    async def list_dir(self, path):
        dirs = await self.mav.ftp.list_directory(path)
        print(dirs)
    async def run(self):
        await self.init_connection()
        await self.list_dir("/")

def main():
    mavftp = MavFtpHandler()
    asyncio.run(mavftp.run())

if __name__ == "__main__":
    main()
```
I am using a px4 connected via USB. I tried running mavsdk_server manually, here's the output:
```
[02:23:51|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[02:23:51|Info ] Waiting to discover system on serial:///dev/ttyACM0... (connection_initiator.h:20)
[02:23:51|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[02:23:51|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[02:23:51|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[02:23:51|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[02:23:51|Info ] System discovered (connection_initiator.h:63)
[02:23:51|Info ] Server started (grpc_server.cpp:53)
[02:23:51|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
```
I am running linux. Is there any way to fix this?

julianoes, 2023-07-09:
What is the other side? Is it PX4, and which version?

FYI: I'm pretty much re-writing FTP for MAVSDK v2 but it's not finished yet.

durka, 2024-05-15:
I am getting the same error trying to use mavftp with Ardupilot 4.4.4.

julianoes, 2024-05-20:
Yes, I see this too. On my list for this week.