mavlink/MAVSDK-Python #613: Timeout when trying to use gimbal

blank-supportgis, 2023-07-17:
I've connected a gimbal to a Pixhawk 4 running PX4 via PWM1. Furthermore, a Raspberry Pi is connected via the TELEM port. My parameter list can be downloaded [here](https://www.dropbox.com/scl/fi/2a8m3sbo4ogsmvuh0vqjv/param_dump_230714.params?rlkey=mvyy9gfcu97zs3wxf1bl6fvzz&dl=0).

When I try to run the [gimbal example](https://github.com/mavlink/MAVSDK-Python/blob/ac985f523cadefbe85ed0524e0b4779392678e9a/examples/gimbal.py) , substituting the SITL address for the physical connection, I get the following exception as soon as any gimbal command is executed:
```
Traceback (most recent call last):
File “gimbal.py”, line 61, in
asyncio.run(run())
File “/usr/lib/python3.7/asyncio/runners.py”, line 43, in run
return loop.run_until_complete(main)
File “/usr/lib/python3.7/asyncio/base_events.py”, line 584, in run_until_complete
return future.result()
File “gimbal.py”, line 23, in run
await drone.gimbal.set_mode(GimbalMode.YAW_FOLLOW)
File “/home/pi/venv_mission/lib/python3.7/site-packages/mavsdk/gimbal.py”, line 484, in set_mode
raise GimbalError(result, “set_mode()”, gimbal_mode)
mavsdk.gimbal.GimbalError: TIMEOUT: ‘Timeout’; origin: set_mode(); params: (<GimbalMode.YAW_FOLLOW: 0>,)
```
The [takeoff_and_land](https://github.com/mavlink/MAVSDK-Python/blob/ac985f523cadefbe85ed0524e0b4779392678e9a/examples/takeoff_and_land.py) script manages to connect to the Pixhawk, so the hardware connection seems to be fine.

JonasVautherin, 2023-07-17:
Are you sure that your gimbal is sending MAVLink messages and that they reach MAVSDK?

blank-supportgis, 2023-07-17:
@JonasVautherin It's not a MAVLink gimbal, it only works via a PWM channel. I used this gimbal before in other builds and the gimbal commands worked even when the gimbal wasn't connected at all.
As for MAVLink communication between MAVSDK and the Pixhawk, I don's see anything in QGroundControl's MAVLink Inspector. Is there another way?

blank-supportgis, 2023-07-17:
I'm afraid, it works now and I haven't got a clue why.
My RPi has an LTE-module, which couldn't connect, because I left the antennas off. When I booted with the antennas on, the LTE connection was properly initialized and from that moment on, the gimbal started working. Don't know if there is any relation, but that is all I can say...

JonasVautherin, 2023-07-17:
Oh... so this can be closed then? :blush:

julianoes, 2023-07-17:
I was going to comment that you might want to consider updating the PX4 version because gimbal support has been improved since v1.12 (what it says in the params file).

But anyway, if it's working now, I'll close this.

closed 2023-07-17 by julianoes