mavlink/MAVSDK-Python #615: I am doing missions by giving setpoints in Offboard mode using local positioning.it is working fine but  unable to move direction, drone moves to all points keeping the nose in one direction. i want to move the drone nose in the direction drone moving. Any help

bb17pugc, 2023-08-28:


JonasVautherin, 2023-08-28:
Can you please format the issue properly? With a title, a description, maybe some code snippet showing your intent, etc? We are not here to format your issue and guess what you are trying to do.

julianoes, 2023-08-28:
@bb17pugc PX4 or ArduPilot? And which offboard function are you using?

bb17pugc, 2023-08-29:
> @bb17pugc PX4 or ArduPilot? And which offboard function are you using?
i am using PX4 and using MAVSDK offboard functions from [this ](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py) to provide the setpoints

julianoes, 2023-08-29:
Aha, and the yaw argument does not work? Which PX4 version?

bb17pugc, 2023-08-29:
> Aha, and the yaw argument does not work? Which PX4 version?

we don't want to give it the static yaw value , we want it to calculate the yaw automatically when moving to next set point 
The details are
px4 version : 1.11.0
pixhawk : 6x
camera for local positioning : Stereolabs camera ZED 2i (providing position using mavros and setpoints 
are being provided using mavsdk)

julianoes, 2023-08-29:
Aha. Have you tried setting yaw to NaN? I think in python that would be `float('nan')`.

bb17pugc, 2023-08-30:
> Aha. Have you tried setting yaw to NaN? I think in python that would be `float('nan')`.

Yes i tried it and got the below message (see in image)
i have attached my source code as well please take a look

[image]

    from mavsdk import System
    from mavsdk.offboard import (OffboardError, PositionNedYaw,Attitude , PositionNedYaw)
    from mavsdk.telemetry import FlightMode
    import asyncio
    
    #mission_waypoints : this is array of set points (each set point is a array of north_m,east_m,down_m,yaw values) recieving from other method and its format is like this:
    #[
         #[north_m , east_m, down_m , yaw]
    #]
    async def run(mission_waypoints):
        drone = System()
        await drone.connect()
        # await drone.connect(system_address='serial:///dev/ttyUSB0:57600')
    
        print("Waiting for drone to connect...")
        async for state in drone.core.connection_state():
            if state.is_connected:
                print(f"-- Connected to drone!")
                break
    
        print("-- Arming")
        await drone.action.arm()
    
        print("-- Setting initial setpoint")
        await drone.offboard.set_position_ned(PositionNedYaw(mission_waypoints[0][0],mission_waypoints[0][1],mission_waypoints[0][2], mission_waypoints[0][3]))
    
        print("-- Starting offboard")
        try:
            await drone.offboard.start()
        except OffboardError as error:
            print(f"Starting offboard mode failed \
                    with error code: {error._result.result}")
            print("-- Disarming")
            await drone.action.disarm()
            return
    
        print_speed_task = asyncio.ensure_future(start_mission(mission_waypoints,drone))
        await asyncio.sleep(10000)
    
        try:
            await drone.offboard.stop()
            await drone.action.land()
            print_speed_task.cancel()
            
        except OffboardError as error:
            print(f"Stopping offboard mode failed \
                    with error code: {error._result.result}")

    async def start_mission(mission_waypoints,drone):
      currentMissionIndex = 0
      print("starting getting drone local position")
      for pos in drone.telemetry.position_velocity_ned():

              if(mission_waypoints[currentMissionIndex][0] == round(pos.position.north_m,1) and  
              mission_waypoints[currentMissionIndex][1] == round(pos.position.east_m,1) and 
              mission_waypoints[currentMissionIndex][2] == round(pos.position.down_m,1)):
                print(mission_waypoints[currentMissionIndex])

                currentMissionIndex = currentMissionIndex+1
                if currentMissionIndex == len(mission_waypoints):        
                    try:
                        print("returning to home")
                        await drone.offboard.stop()
                        await drone.action.return_to_launch()
                        return
                    except OffboardError as error:
                        print(f"Stopping offboard mode failed \
                                with error code: {error._result.result}")
                        return
                
                print("going to way point:")
                print(currentMissionIndex+1)
                
                await drone.offboard.set_position_ned(
                PositionNedYaw(mission_waypoints[currentMissionIndex][0],mission_waypoints[currentMissionIndex][1],mission_waypoints[currentMissionIndex][2], float('nan') ))
  
    #Important Note:
    #Go 0m North, 0m East, 0m Down \
    #Down:moving downward if positive value given , moves upward if negative value given
    #North:moving forward if positive value given , moves backward if negative value given
    #East : moves right if positive value is given , moves left if negative value is given`

julianoes, 2023-08-30:
Ok, so it sounds like PX4 doesn't like NAN for yaw then.

Would it be a possibility to create the yaw trajectory that you want in your Python program and send it in small increments?

bb17pugc, 2023-09-01:
> Ok, so it sounds like PX4 doesn't like NAN for yaw then.
> 
> Would it be a possibility to create the yaw trajectory that you want in your Python program and send it in small increments?

It is very difficult for me to create the yaw trajectory so any easy solution can you  suggest Please?

julianoes, 2023-09-01:
You want yaw to be from the last point to the next point, right? So then you should be able to work out the yaw angle using trigonometry, presumably using atan2.

pd3a, 2023-09-16:
Hi, Does guided mode support exist for MAVSDK to run on ArduPilot?