mavlink/MAVSDK-Python #618: examples/manual_control.py file  does not run

alanmehio, 2023-09-12:
The line needs to be change 
from 
asyncio.run(run())
to 
asyncio.run(manual_controls())

Also, there is a error once this file get executed under the SITL ( jmavsim simulation ) as below

**mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()**

Steps to reproduce:
1- run the jmavsim simulation  from command line as below
make px4_sitl_default jmavsim
2- from IDE  run  manual_control.py

JonasVautherin, 2023-09-13:
Would you like to open a pull request? :blush:

alanmehio, 2023-09-13:
Yes, I will do that.
Cheers
Alan

alanmehio, 2023-09-13:
This issue is closed now since the pull request is created.

closed 2023-09-13 by alanmehio

JonasVautherin, 2023-09-13:
Nitpick: the way it works usually is that we close the issue after the PR is merged ("the PR closes the issue").

alanmehio, 2023-09-13:
This has been noted
Cheers
Alan