mavlink/MAVSDK-Python #621: from mavsdk import System

lankoestee, 2023-09-20:
I use `pip3 install mavsdk` to install mavsdk. I want to use mavsdk to control the actuator for throwing, so I change the example `takeoff_and_land.py` to the following code.
```python
#!/usr/bin/env python3
# set_actuator.py

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="udp://:50051")

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    # print("Waiting for drone to have a global position estimate...")
    # async for health in drone.telemetry.health():
    #     if health.is_global_position_ok and health.is_home_position_ok:
    #         print("-- Global position estimate OK")
    #         break

    print("-- Arming")
    await drone.action.arm()

    print("-- Set actuator control")
    await drone.action.set_actuator(1, 0.5)

    # print("-- Taking off")
    # await drone.action.takeoff()

    # await asyncio.sleep(10)

    # print("-- Landing")
    # await drone.action.land()

    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
```
I should be able to use this code to control the actuator, right？
I had used QGC to set the actuator to `offboard actuator set 1` and it can move through Actuator Testing.
Then I connect the flight control of px4 hardware through USB. And I start `roslaunch mavros px4.launch` to test it. But when I run `python3 set_actuator.py`, I got `ImportError: cannot import name 'System' from 'mavsdk'` 
What should I do? Am I following the correct steps？

julianoes, 2023-09-21:
> ImportError: cannot import name 'System' from 'mavsdk'

Have you installed mavsdk using pip?

And by the way, if you're using ROS already, you might just be able to do everything with ROS and MAVROS and you don't actually need MAVSDK.

Jai-GAY, 2023-10-23:
I got the following error [trying out the modified takeoff_and_land.py example](https://github.com/mavlink/MAVSDK-Python/blob/cd941eb6dacd297c4ec5567e9a20ce5fcd3dfde0/examples/takeoff_and_land.py), but got the following error with Arducopter using Pi4B.

Waiting for drone to connect...
-- Connected to drone!
Version: [flight_sw_major: 4, flight_sw_minor: 3, flight_sw_patch: 0, flight_sw_vendor_major: 52, flight_sw_vendor_minor: 51, flight_sw_vendor_patch: 57, os_sw_major: 0, os_sw_minor: 0, os_sw_patch: 0, flight_sw_git_hash: 313762383, os_sw_git_hash: 643065643565363]
exception calling callback for <Future at 0xf46ec838 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/pi/.local/lib/python3.9/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/pi/.local/lib/python3.9/site-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/home/pi/.local/lib/python3.9/site-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.9/concurrent/futures/_base.py", line 329, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.9/asyncio/futures.py", line 398, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.9/asyncio/base_events.py", line 791, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.9/asyncio/base_events.py", line 510, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed

```
async def run():
    drone = System()
    await drone.connect(system_address="serial:///dev/ttyAMA1:921600")

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    info = await drone.info.get_version()
    print(info)

    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
```

julianoes, 2023-10-23:
@Jai-GAY please don't just ask your question in multiple threads. Create a new one next time, thank yo.

Jai-GAY, 2023-10-24:
> @Jai-GAY please don't just ask your question in multiple threads. Create a new one next time, thank you.

I did not know they are related. I see they are independent problem.

closed 2024-06-17 by lankoestee