mavlink/MAVSDK-Python #626: Error getting started with mavsdk package

camchaney, 2023-09-29:
I can't seem get the mavsdk package started on my Raspberry Pi 4. I installed using `pip3 install mavsdk` and went to test it out using `apython`:
`from mavsdk import System`
This throws the error:
`ImportError: /lib/arm-linux-gnueabihf/libstdc++.so.6: version `GLIBCXX_3.4.29' not found`
I tried it with `python` too and same results.

I tried a couple things to get around this including reinstalling certain packages via a suggestion from another user:
`pip3 uninstall -y grpcio grpcio-tools
sudo apt install -y python3-grpcio python3-grpc-tools`
This gives another error `ModuleNotFoundError: No module named 'grpc'`.

How can I fix this and get mavsdk to work?

JonasVautherin, 2023-09-30:
Which version is pip trying to pull? Is it really the last one? I thought we were bundling the musl version with MAVSDK-Python, not glibc anymore (see [here](https://github.com/mavlink/MAVSDK-Python/blob/main/setup.py#L61)) :thinking:.

Does [the arm64 mavsdk_server from the release page](https://github.com/mavlink/MAVSDK/releases/download/v1.4.17/mavsdk_server_linux-arm64-musl) run on your RPi 4?

julianoes, 2023-10-05:
@camchaney yes, please post the output of `pip3 install --upgrade mavsdk`