mavlink/MAVSDK-Python #628: sending custom message from companion computer to pixhawk to recieve in GCS

arpitpara, 2023-10-13:
Hi
I was wondering that is it possible to send some custom message strings through mavsdk-python from companion computer to pixhawk to recieve it in GCS.
For example, some sensor data attached to Companion computer or some feedback messages.

julianoes, 2023-10-16:
This is currently not possible. May I ask about the information/intent of that message? Maybe we can find an alternative.

arpitpara, 2023-10-16:
I am using Flight controller arm/disarm values to trigger some scripts in my companion computer. I am able to do it sucessfully, but I don't get any feedback messages from my companion computer in my GCS that scripts are working fine. So I thought of sending some custom strings to Flight controller so I can map those values in GCS to represent some particaular messages.