mavlink/MAVSDK-Python #630: Jetson Orin NX Ubuntu 20.04 Python 3.8.x pip install error

Jai-GAY, 2023-10-26:
Does anyone face the same issue?
Are there packages missing that cause this error? do i have to use [python3 -m pip install --upgrade mavsdk](https://github.com/mavlink/MAVSDK-Python/pull/629)?
Currently, this board does not install mavsdk before.

Python 3.8.10
pip 20.0.2 from /usr/lib/python3/dist-packages/pip (python 3.8)
Distributor ID: Ubuntu
Description: Ubuntu 20.04.6 LTS
Release: 20.04
Codename: focal

pip3 install mavsdk

Collecting mavsdk
Downloading mavsdk-1.4.9-py3-none-manylinux2014_aarch64.whl (12.0 MB)
Collecting aiogrpc>=1.8
Downloading aiogrpc-1.8-py3-none-any.whl (10 kB)
Collecting grpcio>=1.50.0
Downloading grpcio-1.59.0.tar.gz (24.8 MB)
**ERROR: Command errored out with exit status 1:**
command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-vfc82mh_/grpcio/setup.py'"'"'; file='"'"'/tmp/pip-install-vfc82mh_/grpcio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(file);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, file, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-vfc82mh_/grpcio/pip-egg-info
cwd: /tmp/pip-install-vfc82mh_/grpcio/
Complete output (9 lines):
Traceback (most recent call last):
File "", line 1, in
File "/tmp/pip-install-vfc82mh_/grpcio/setup.py", line 95, in
import commands
File "/tmp/pip-install-vfc82mh_/grpcio/src/python/grpcio/commands.py", line 31, in
import support
File "/tmp/pip-install-vfc82mh_/grpcio/src/python/grpcio/support.py", line 111, in
errors.CompileError: diagnose_compile_error,
AttributeError: module 'setuptools.errors' has no attribute 'CompileError'
----------------------------------------
**ERROR**: Command errored out with exit status 1: **python setup.py egg_info Check the logs for full command** output.

julianoes, 2023-10-26:
Looks like this: https://github.com/grpc/grpc/issues/34762

Can you try to upgrade setuptools first?

Alternatively, try installing the dependency using `sudo apt install python3-grpcio`.

Jai-GAY, 2023-10-31:
Still getting error after upgrade setuptools from 45.2.0 to 52.0.0 (Pi 4B Bullseye Python3 3.9 was okay)

pip show setuptools
Name: setuptools
Version: 45.2.0

```
Collecting mavsdk
  Using cached mavsdk-1.4.9-py3-none-manylinux2014_aarch64.whl (12.0 MB)
Collecting grpcio>=1.50.0
  Downloading grpcio-1.59.2.tar.gz (24.8 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-h2vdwosx/grpcio/setup.py'"'"'; __file__='"'"'/tmp/pip-install-h2vdwosx/grpcio/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-h2vdwosx/grpcio/pip-egg-info
         cwd: /tmp/pip-install-h2vdwosx/grpcio/
    Complete output (9 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-h2vdwosx/grpcio/setup.py", line 95, in <module>
        import commands
      File "/tmp/pip-install-h2vdwosx/grpcio/src/python/grpcio/commands.py", line 31, in <module>
        import support
      File "/tmp/pip-install-h2vdwosx/grpcio/src/python/grpcio/support.py", line 111, in <module>
        errors.CompileError: diagnose_compile_error,
    AttributeError: module 'setuptools.errors' has no attribute 'CompileError'
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
```

Collecting setuptools==52.0.0
  Downloading setuptools-52.0.0-py3-none-any.whl (784 kB)
ERROR: launchpadlib 1.10.13 requires testresources, which is not installed.
Installing collected packages: setuptools
Successfully installed setuptools-52.0.0

Jai-GAY, 2023-10-31:
this solved the mavsdk installation.

python -m pip install pip==20.3.4

closed 2023-10-31 by Jai-GAY

julianoes, 2023-11-01:
Nice, thanks for the notes!