mavlink/MAVSDK-Python #631: Trouble getting started on Windows

HenryInUtah, 2023-10-29:
Hello,
I am having trouble getting started...

The computer is a Latte Panda running Windows IOT.  I installed mavsdk using pip in the "normal" way without problems.  When I ran one of the example Python scripts it said that the server was not installed.  I installed the first Windows 64 server that I found from June 6 release 1.4.16  Then I installed the missing MSVCP140.dll and it runs.

The desktop shortcut looks like this:
```
C:\Users\LattePanda\Downloads\mavsdk-windows-x64-release\bin\mavsdk_server_bin.exe -p 50051
```
and the information in the DOS window looks like this:
```
[12:47:40|Info ] MAVSDK version: v1.4.16 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:20)
[12:47:40|Info ] Waiting to discover system on udp://:14540... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
```
I changed the example "takeoff_and_land.py" script to this:
```
print('Creating the drone object')
drone = System(mavsdk_server_address='localhost', port=50051)
print('Trying to connect')
await drone.connect(system_address="serial://COM6:921600")
print('Connected')
```
but it never gets past the drone.connect() line.

The connection is via a USB to serial cable plugged into the TELEM2 port on a PixHawk 4.  The cable shows up as COM6 in Windows device manager and the TELEM2 port was working with pymavlink on a Raspberry Pi.  That doesn't mean that everything is perfect, but it should be in the ballpark.  I did swap the Tx and Rx pins on the cable.

Could someone please offer some suggestions of what I am doing wrong?  Is the server and Python code working as expected?  What else can I test?

Thank you for your help.

julianoes, 2023-10-30:
I'm not familiar with serial on Windows.

I suggest to follow these steps:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

Basically, can you start mavsdk_server directly with the serial CLI arg, and then you just use this in Python:

```
drone = System()
await drone.connect()
```

HenryInUtah, 2023-10-30:
Hello Julian,
I followed your suggestions and have both good and not good results.

I started the server manually with a command line parameter pointing towards the serial port and it seems to have connected and be fairly happy
```
C:\Users\LattePanda\Downloads\mavsdk-windows-x64-release\bin>mavsdk_server_bin.exe serial://COM7:921600
[09:56:20|Info ] MAVSDK version: v1.4.16 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:20)
[09:56:20|Info ] Waiting to discover system on serial://COM7:921600... (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:20)
[09:56:20|Debug] New: System ID: 1 Comp ID: 1 (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\mavsdk_impl.cpp:496)
[09:56:20|Debug] Component Autopilot (1) added. (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:377)
[09:56:21|Warn ] Vehicle type changed (new type: 2, old type: 0) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:225)
[09:56:21|Debug] Discovered 1 component(s) (D:\a\MAVSDK\MAVSDK\src\mavsdk\core\system_impl.cpp:578)
[09:56:21|Info ] System discovered (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\connection_initiator.h:63)
[09:56:21|Info ] Server started (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:53)
[09:56:21|Info ] Server set to listen on 0.0.0.0:50051 (D:\a\MAVSDK\MAVSDK\src\mavsdk_server\src\grpc_server.cpp:54)
```
This is literally the first glimmer of hope that I have seen from this system in several days.  It is very encouraging.

Next, I tried to run this simple program:
```
from mavsdk import System
import asyncio

async def run():

	print('Creating the drone object')
	drone = System()
	print('Trying to connect')
	await drone.connect()
	print('Connected')

if __name__ == "__main__":
	# Run the asyncio loop
	asyncio.run(run())
```
and the response is not great:
```
C:\Users\LattePanda\Documents\MavSDK Examples\examples>python Henry.py
Creating the drone object
Trying to connect

This installation does not provide an embedded 'mavsdk_server' binary.
If you installed using pip, this means that 'mavsdk_server' is not distributed
for your platform yet (e.g. arm).

You will need to get and run the 'mavsdk_server' binary manually:

  1. Download 'mavsdk_server' from: https://github.com/mavlink/mavsdk/releases
     or build it from source.

  2. Run it, e.g. on port 50051:
     './mavsdk_server -p 50051'

  3. Set the 'mavsdk_server_address' and port when creating the System:
     'drone = System(mavsdk_server_address='localhost', port=50051)'
```

It seems that the Python module is not finding the server, correct?  The messages from the server indicate that it is running and ready for connections (I think).

Could you suggest what I should investigate next?

Thanks again for your time.

julianoes, 2023-10-30:
Is this and amd64 or arm64 architecture?

HenryInUtah, 2023-10-30:
We tried the following code and it connects to the server.  It seems that you have to specify the default values.

```
from mavsdk import System
import asyncio

async def run():

	print('Creating the drone object')
	drone = System(mavsdk_server_address='localhost', port=50051)
	print('Trying to connect')
	await drone.connect()
	print('Connected')

if __name__ == "__main__":
	# Run the asyncio loop
	asyncio.run(run())
```

The next problem is that it only connects to the TELEM2 port on the PX4 and not on TELEM1.  We need TELEM2 for something else.  Supposedly, both ports are set as MavLink ports but there must be something else that we missed.  More time on Google.

Thank you very much for your help and suggestions.  We aren't running quite yet, but getting closer.

julianoes, 2023-11-01:
Regarding Telem2 vs. Telem1. Check the flow control settings and try force on or force off depending on whether you have it connected or not.

closed 2023-11-01 by julianoes