mavlink/MAVSDK-Python #632: get_version raise InfoError

Jai-GAY, 2023-10-31:
Distributor ID:	Ubuntu
Description:	Ubuntu 20.04.5 LTS
Python 3.8.10
Name: mavsdk
Version: 1.4.9

```
System server b4
System server done
Waiting for drone to connect...
-- Connected to drone!
Traceback (most recent call last):
  File "mavsdk_test_2.py", line 75, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mavsdk_test_2.py", line 24, in run
    info = await drone.info.get_version()
  File "/home/orinnx-1of3/.local/lib/python3.8/site-packages/mavsdk/info.py", line 746, in get_version
    raise InfoError(result, "get_version()")
mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET: 'Information Not Received Yet'; origin: get_version(); params: ()
```

```
async def run():
    # Init the drone
    #UART 4
    print(f"System server b4")
    drone = System()
    print(f"System server done")
    await drone.connect(system_address="serial:///dev/ttyTHS0:921600")

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f"-- Connected to drone!")
            break

    info = await drone.info.get_version()
    print(info)

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
```

JonasVautherin, 2023-10-31:
> INFORMATION_NOT_RECEIVED_YET

MAVSDK returns this error until it receives the `MAVLINK_MSG_ID_AUTOPILOT_VERSION` message ([see here](https://github.com/mavlink/MAVSDK/blob/b29239df34afa4bfc227b73e98d06c28c264dfa7/src/mavsdk/plugins/info/info_impl.cpp#L154C4-L154C4)). Note that `get_version()` is implemented [here](https://github.com/mavlink/MAVSDK/blob/b29239df34afa4bfc227b73e98d06c28c264dfa7/src/mavsdk/plugins/info/info_impl.cpp#L229).

To me it sounds like your script is requesting the version info before MAVSDK has received it.

You can try multiple things, I would say:

1. Have your script wait longer (maybe catch the exception and try again for a while).
2. Check that your drone sends the `MAVLINK_MSG_ID_AUTOPILOT_VERSION` message indeed. But since you mention that it works on the RPi, if that's the same autopilot, then I would expect it to behave similarly.

Jai-GAY, 2023-11-01:
> it works on the RPi, if that's the same autopilot, then I would expect it to behave similarly.

Yes, it works with Pi running python 3.9.x
I will give a few different tests again to see if timing is the issue.

Jai-GAY, 2023-11-02:
I thought I run the python script early while Arducopter is not ready, new test sequence, arducopter ready then run the python script. more error information. no different if i add a delay.

```
    await asyncio.sleep(2)

    info = await drone.info.get_version()
    print(info)
```
```
System server b4
System server done
Waiting for drone to connect...
-- Connected to drone!
exception calling callback for <Future at 0xffffa429d610 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.CANCELLED
	details = "Locally cancelled by application!"
	debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
Traceback (most recent call last):
  File "mavsdk_test_2.py", line 75, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mavsdk_test_2.py", line 24, in run
    info = await drone.info.get_version()
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/mavsdk/info.py", line 746, in get_version
    raise InfoError(result, "get_version()")
mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET: 'Information Not Received Yet'; origin: get_version(); params: ()
```

JonasVautherin, 2023-11-02:
When does Ardupilot send `MAVLINK_MSG_ID_AUTOPILOT_VERSION`? Would be interesting to know that. Like does it always broadcast it, do you have to `set_rate`, do you have to request it somehow?

Jai-GAY, 2023-11-03:
The [identical code that work on Pi4B](https://github.com/mavlink/MAVSDK-Python/issues/453#issuecomment-1790270693) does not has the same results on Orin-NX. I suspect may be due to [different asyncio version](https://pypi.org/project/asyncio/) where I cannot or know how to verify. I also do not know how was it get installed, when asyncio was first installed.
https://docs.python.org/3/library/asyncio.html

```
System server b4
System server done
Orin-NX Waiting for drone to connect...
-- Connected to drone!
print_status_text CancelledError
exception calling callback for <Future at 0xffff9c15e970 state=finished raised _MultiThreadedRendezvous>
Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/thread.py", line 57, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
        status = StatusCode.CANCELLED
        details = "Locally cancelled by application!"
        debug_error_string = "None"
>

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.8/concurrent/futures/_base.py", line 328, in _invoke_callbacks
    callback(self)
  File "/usr/lib/python3.8/asyncio/futures.py", line 374, in _call_set_state
    dest_loop.call_soon_threadsafe(_set_state, destination, source)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 764, in call_soon_threadsafe
    self._check_closed()
  File "/usr/lib/python3.8/asyncio/base_events.py", line 508, in _check_closed
    raise RuntimeError('Event loop is closed')
RuntimeError: Event loop is closed
Traceback (most recent call last):
  File "mavsdk_test_20231102_3.py", line 90, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "mavsdk_test_20231102_3.py", line 26, in run
    info = await drone.info.get_version()
  File "/home/orinnx-3of3/.local/lib/python3.8/site-packages/mavsdk/info.py", line 746, in get_version
    raise InfoError(result, "get_version()")
mavsdk.info.InfoError: INFORMATION_NOT_RECEIVED_YET: 'Information Not Received Yet'; origin: get_version(); params: ()
```

JonasVautherin, 2023-11-03:
The error you receive ("INFORMATION_NOT_RECEIVED_YET") comes from MAVSDK in C++. I don't think it is an asyncio issue.

Really I would check (maybe with Wireshark) if `MAVLINK_MSG_ID_AUTOPILOT_VERSION` is sent by the autopilot and received by MAVSDK.

Jai-GAY, 2023-11-06:
I do not suspect the cable got an issue as [Mavproxy](https://ardupilot.org/mavproxy/index.html) and [uart_example.py](https://github.com/JetsonHacksNano/UARTDemo) look good.

```
mavproxy.py --master=/dev/ttyTHS0,921600
WARNING: You should uninstall ModemManager as it conflicts with APM and Pixhawk
Connect /dev/ttyTHS0,921600 source_system=255
Log Directory:
Telemetry log: mav.tlog
Waiting for heartbeat from /dev/ttyTHS0
MAV> Detected vehicle 1:1 on link 0
online system 1
STABILIZE> Mode STABILIZE

STABILIZE> status
STABILIZE> set requireexit True
Counters: MasterIn:[283] MasterOut:255 FGearIn:0 FGearOut:0 Slave:0
MAV Errors: 0
None
257: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
26: TIMESYNC {tc1 : 0, ts1 : 310726049001}
STABILIZE> status
STABILIZE> time
Counters: MasterIn:[306] MasterOut:276 FGearIn:0 FGearOut:0 Slave:0
MAV Errors: 0
None
278: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
28: TIMESYNC {tc1 : 0, ts1 : 330731412001}
STABILIZE> status
No SYSTEM_TIME time available
STABILIZE> status
Counters: MasterIn:[1226] MasterOut:1109 FGearIn:0 FGearOut:0 Slave:0
MAV Errors: 0
None
1115: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
111: TIMESYNC {tc1 : 0, ts1 : 1160926338001}
STABILIZE> exit
Counters: MasterIn:[1236] MasterOut:1117 FGearIn:0 FGearOut:0 Slave:0
MAV Errors: 0
None
1124: HEARTBEAT {type : 2, autopilot : 3, base_mode : 81, custom_mode : 0, system_status : 3, mavlink_version : 3}
112: TIMESYNC {tc1 : 0, ts1 : 1170928846001}
STABILIZE> exit()
Unloading module link
Unloading module log
Unloading module signing
Unloading module wp
Unloading module rally
Unloading module fence
Unloading module ftp
Unloading module param
Unloading module relay
Unloading module tuneopt
Unloading module arm
Unloading module mode
Unloading module calibration
Unloading module rc
Unloading module auxopt
Unloading module misc
Unloading module cmdlong
Unloading module battery
Unloading module terrain
Unloading module output
Unloading module adsb
Unloading module layout
Exception in thread log_writer:
Traceback (most recent call last):
  File "/usr/lib/python3.8/threading.py", line 932, in _bootstrap_inner
```
[image]

JonasVautherin, 2023-11-06:
> I do not suspect the cable got an issue as [Mavproxy](https://ardupilot.org/mavproxy/index.html) and [uart_example.py](https://github.com/JetsonHacksNano/UARTDemo) look good.

Sorry where do you see that `MAVLINK_MSG_ID_AUTOPILOT_VERSION` is being received here? I only see `HEARTBEAT` and `TIMESYNC` :thinking:.

Jai-GAY, 2023-11-06:
> where do you see that `MAVLINK_MSG_ID_AUTOPILOT_VERSION` is being received

[Here](https://github.com/mavlink/MAVSDK-Python/issues/453#issuecomment-1774507019).

Jai-GAY, 2024-02-08:
$ pip3 install mavsdk==2.0.0 --upgrade
ERROR: Could not find a version that satisfies the requirement mavsdk==2.0.0 (from versions: 0.5.2, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.12.0, 0.13.4, 0.14.0, 0.15.0, 0.16.0, 0.16.1, 0.17.0, 0.18.0, 0.19.0, 0.20.0, 0.21.0, 0.22.0, 0.22.1, 0.22.2, 0.22.3, 0.23.0, 1.0.0, 1.2.0, 1.3.0, 1.4.0, 1.4.1, 1.4.2, 1.4.3, 1.4.4, 1.4.5, 1.4.6, 1.4.7, 1.4.8, 1.4.9)
ERROR: No matching distribution found for mavsdk==2.0.0

$ sudo lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 20.04.6 LTS
Release:        20.04
Codename:       focal

julianoes, 2024-02-08:
Please create a new issue @Jai-GAY and provide all necessary information/context.

closed 2024-02-08 by julianoes