mavlink/MAVSDK-Python #633: goto example returns action error

yunpengye, 2023-11-01:
I tried the `goto.py` in the examples but returns a action error

Tried in SITL Gazebo
```
python3 goto.py 
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...
-- Global position state is good enough for flying.
Fetching amsl altitude at home location....
-- Arming
-- Taking off
Traceback (most recent call last):
  File "goto.py", line 47, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.8/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.8/asyncio/base_events.py", line 616, in run_until_complete
    return future.result()
  File "goto.py", line 38, in run
    await drone.action.goto_location(47.397606, 8.543060, flying_alt, 0)
  File "/home/yunpeng/.local/lib/python3.8/site-packages/mavsdk/action.py", line 548, in goto_location
    raise ActionError(result, "goto_location()", latitude_deg, longitude_deg, absolute_altitude_m, yaw_deg)
mavsdk.action.ActionError: UNSUPPORTED: 'Unsupported'; origin: goto_location(); params: (47.397606, 8.54306, 508.1050109863281, 0)
```

Jaeyoung-Lim, 2023-11-01:
I can confirm that this is a problem. The vehicle actually reaches the target point, but mavsdk returns an error

@JonasVautherin @julianoes Any clues on how we can solve this?

JonasVautherin, 2023-11-01:
Seems like it throws [`UNSUPPORTED`](https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/action/action.proto#L315), which [from the implementation](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L440) seems to come either from [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L461) or from [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L476), which is translated from [here](https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L790-L791) :thinking:.

Could it be that PX4 returned [MAV_RESULT_UNSUPPORTED](https://mavlink.io/en/messages/common.html#MAV_RESULT_UNSUPPORTED) for some reason?

Jaeyoung-Lim, 2023-11-02:
@JonasVautherin Looked a bit into PX4 side, and indeed PX4 returns `MAV_RESULT_UNSUPPORTED`

However, it is complaining that the mode switch was not requested: https://github.com/PX4/PX4-Autopilot/blob/d6dbf38a1ba376cde9e65c10430e6feb52f6fdd5/src/modules/commander/Commander.cpp#L723

Does this need to be something handled on the mavsdk action plugin side?

Jaeyoung-Lim, 2023-11-02:
@JonasVautherin I have made a fix here: https://github.com/mavlink/MAVSDK/pull/2170 but kept it as a draft since I was not able to test it

farhad-ibrahimzade, 2023-11-19:
Hey all, thank you for opening this issue. I seem to be getting the same result when I run goto.py in examples. I have looked at the fix [here](https://github.com/mavlink/MAVSDK/pull/2170), but I am not sure how to proceed to fix the error. Can you please explain how to implement the fix in MAVSDK Python?

farhad-ibrahimzade, 2023-11-21:
> @JonasVautherin I have made a fix here: [mavlink/MAVSDK#2170](https://github.com/mavlink/MAVSDK/pull/2170)

Thank you for this. Could you please walk me through how to implement the fix in mavsdk python?

julianoes, 2023-11-22:
So the fix is in main, and not the current released v1.4 MAVSDK version. So I assume we have to wait until v2 is released?