mavlink/MAVSDK-Python #636: Offboard mode cannot turn on ERROR: NO_SETPOINT_SET

cameronpm1, 2023-11-15:
I am trying to set up my drone in offboard mode using a pixhawk 6C and Jetson Nano. Everything so far is working, except offboard mode throws out an error no_setpoint_set when I try to turn it on. I attached my code as well as outputs below, I am essentially following the example code provided on git. Does anyone have any familiarity with this problem? Thank you

Terminal output:

Waiting for connection...
-- connection successful
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Setting initial setpoint
-- Starting offboard
Starting offboard mode failed                     with error code: NO_SETPOINT_SET
-- Disarming

Code:

async def demo1(
        self,
        port: str = 'serial:///dev/ttyTHS1:57600',
    ):

        await self.drone.connect(system_address=port)

        print("Waiting for connection...")
        async for state in self.drone.core.connection_state():
            if state.is_connected:
                print(f"-- connection successful")
                break

        print("Waiting for drone to have a global position estimate...")
        async for health in self.drone.telemetry.health():
            if health.is_global_position_ok and health.is_home_position_ok:
                print("-- Global position estimate OK")
                break

        await asyncio.sleep(1)

        print("-- Arming")
        await self.drone.action.arm()

        await asyncio.sleep(0.5)

        print("-- Setting initial setpoint")
        await self.drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))

        await asyncio.sleep(0.5)

        print("-- Starting offboard")
        try:
            await self.drone.offboard.start()
        except OffboardError as error:
            print(f"Starting offboard mode failed \
                    with error code: {error._result.result}")
            print("-- Disarming")
            await self.drone.action.disarm()
            return

julianoes, 2023-11-15:
Can you try without the sleep before `offboard.start()`?

cameronpm1, 2023-11-16:
yes, I removed the sleep commands, same error but I get a little more ouput:

Waiting for connection...
-- connection successful
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
-- Setting initial setpoint
-- Starting offboard
Starting offboard mode failed                     with error code: NO_SETPOINT_SET
-- Disarming
Traceback (most recent call last):
  File "envs/drone_env.py", line 315, in demo1
    await self.drone.offboard.start()
  File "/usr/local/lib/python3.7/dist-packages/mavsdk/offboard.py", line 1150, in start
    raise OffboardError(result, "start()")
mavsdk.offboard.OffboardError: NO_SETPOINT_SET: 'No Setpoint Set'; origin: start(); params: ()

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "envs/drone_env.py", line 378, in <module>
    asyncio.run(env.demo1())
  File "/usr/lib/python3.7/asyncio/runners.py", line 43, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.7/asyncio/base_events.py", line 579, in run_until_complete
    return future.result()
  File "envs/drone_env.py", line 320, in demo1
    await self.drone.action.disarm()
  File "/usr/local/lib/python3.7/dist-packages/mavsdk/action.py", line 343, in disarm
    raise ActionError(result, "disarm()")
mavsdk.action.ActionError: COMMAND_DENIED_LANDED_STATE_UNKNOWN: 'Command Denied Landed State Unknown'; origin: disarm(); params: ()

cameronpm1, 2023-11-16:
Is it possible I have to change some of the parameters on the px4?

julianoes, 2023-11-16:
What's your mavsdk version?

Do ```python -m pip install --upgrade mavsdk``` and show the output.

cameronpm1, 2023-11-16:
this is what I get:

Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: mavsdk in /usr/local/lib/python3.7/dist-packages (1.4.9)
Requirement already satisfied: aiogrpc>=1.8 in /usr/local/lib/python3.7/dist-packages (from mavsdk) (1.8)
Requirement already satisfied: grpcio>=1.50.0 in /usr/local/lib/python3.7/dist-packages (from mavsdk) (1.59.0)
Requirement already satisfied: protobuf<=3.20.1 in /usr/local/lib/python3.7/dist-packages (from mavsdk) (3.20.1)

julianoes, 2023-11-18:
Ok, so that's latest.

Let me try the offboard example against SITL to check.

julianoes, 2023-11-18:
Ok, so these steps worked for me with SITL:
```
python3 -m pip install --upgrade mavsdk
cd examples
./offboard_position_ned.py
```

Can you run the mavsdk_server separate to see what is going on?

It's explained here:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

cameronpm1, 2023-11-20:
Yes, I will try that tomorrow. I have found that if I ran the takeoff command mavsdk then allows me to engage offboard mode. But, I have yet to test if the drone will the move to the desired NED coordinates using set_position, so I am not sure if this is actually a fix or not

cameronpm1, 2023-11-20:
when I ran the file in a separate terminal while running my code to connect with the drone it never recognized that a new connection was found. This is what I got:

usr/local/lib/python3.7/dist-packages/mavsdk/bin/mavsdk_server udp://:14540
[02:15:21|Info ] MAVSDK version: v1.4.17 (mavsdk_impl.cpp:20)
[02:15:21|Info ] Waiting to discover system on udp://:14540... (connection_initiator.h:20)

julianoes, 2023-11-20:
>  I did notice that the mavsdk is installed in my usr file, not .local, could this be the reason it is not working?

No, that's just dependent on how you install pip packages. Same thing.

But it looks like it doesn't actually connect.

However, you're using udp://:14540 in the snippet above but you're probably meaning to use `serial:///dev/ttyTHS1:57600` like in the first post.

cameronpm1, 2023-11-21:
Ah, that makes a lot more sense, reading the docs I was confused whether I was supposed to keep the connection local or the uart port connected to the pixhawk. Anyway, I think we got a little closer, seems there was an error after connecting. Here's what I got:

magpie1@magpie1:/$ sudo usr/local/lib/python3.7/dist-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyTHS1:57600
[08:16:18|Info ] MAVSDK version: v1.4.17 (mavsdk_impl.cpp:20)
[08:16:18|Info ] Waiting to discover system on serial:///dev/ttyTHS1:57600... (connection_initiator.h:20)
[08:16:18|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[08:16:18|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[08:16:21|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[08:16:21|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[08:16:21|Info ] System discovered (connection_initiator.h:63)
E1120 20:16:21.499319633   14770 server_chttp2.cc:49]        {"created":"@1700529381.498788019","description":"No address added out of total 1 resolved","file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/ext/transport/chttp2/server/chttp2_server.cc","file_line":872,"referenced_errors":[{"created":"@1700529381.498733748","description":"Failed to add any wildcard listeners","file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/tcp_server_posix.cc","file_line":346,"referenced_errors":[{"created":"@1700529381.498648123","description":"Unable to configure socket","fd":6,"file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/tcp_server_utils_posix_common.cc","file_line":217,"referenced_errors":[{"created":"@1700529381.498624165","description":"Address in use","errno":98,"file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/tcp_server_utils_posix_common.cc","file_line":196,"os_error":"Address in use","syscall":"listen"}]},{"created":"@1700529381.498730988","description":"Unable to configure socket","fd":6,"file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/tcp_server_utils_posix_common.cc","file_line":217,"referenced_errors":[{"created":"@1700529381.498712446","description":"Address in use","errno":98,"file":"/work/build/linux-arm64-musl/third_party/grpc/grpc/src/grpc/src/core/lib/iomgr/tcp_server_utils_posix_common.cc","file_line":196,"os_error":"Address in use","syscall":"listen"}]}]}]}
[08:16:21|Error] Failed to bind server to port 50051 (grpc_server.cpp:56)
Failed to start, exiting...

cameronpm1, 2023-11-22:
That was on my own adapted example, I had to take my work home for the holidays, and now everything seems to run normal. This is what I get when I run the takeoff and land example:

magpie1@magpie1:/$ sudo usr/local/lib/python3.7/dist-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyTHS1:57600
[sudo] password for magpie1: 
[05:17:36|Info ] MAVSDK version: v1.4.17 (mavsdk_impl.cpp:20)
[05:17:36|Info ] Waiting to discover system on serial:///dev/ttyTHS1:57600... (connection_initiator.h:20)
[05:17:36|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[05:17:36|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[05:17:36|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[05:17:36|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[05:17:36|Info ] System discovered (connection_initiator.h:63)
[05:17:36|Info ] Server started (grpc_server.cpp:53)
[05:17:36|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[05:17:53|Debug] MAVLink: info: Armed by external command        (system_impl.cpp:250)
[05:17:54|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-11-22/20_16_41.ulg         (system_impl.cpp:250)
[05:17:54|Debug] MAVLink: info: Using minimum takeoff altitude: 5.00 m   (system_impl.cpp:250)
[05:17:55|Debug] MAVLink: info: Takeoff detected         (system_impl.cpp:250)
[05:17:59|Debug] MAVLink: emergency: Preflight Fail: low battery         (system_impl.cpp:250)
[05:17:59|Debug] MAVLink: critical: Failsafe activated   (system_impl.cpp:250)
[05:18:00|Debug] MAVLink: critical: Failsafe activated   (system_impl.cpp:250)
[05:18:00|Debug] MAVLink: critical: Failsafe activated   (system_impl.cpp:250)
[05:18:01|Debug] MAVLink: emergency: Preflight Fail: low battery         (system_impl.cpp:250)
[05:18:03|Debug] MAVLink: info: Landing at current position      (system_impl.cpp:250)
[05:18:09|Debug] MAVLink: info: Landing detected         (system_impl.cpp:250)
[05:18:11|Debug] MAVLink: info: Disarmed by landing      (system_impl.cpp:250)

The only thing that has changed with the drone is the gps lock is much stronger here. Even though my gps was not failing while running these tests is it possible that my issues were all because the gps was losing signal for ver short periods of time and I was unaware?

julianoes, 2023-11-23:
I think it was either a MAVSDK version problem or timing with the sleeps.