mavlink/MAVSDK-Python #637: telemetry.py only reports position data and eventually fails

maciek01, 2023-11-19:
I am runing the telemetry.oy example against arducopter 4.2.2 FC.

The example only reports position telementry. I does not report neither gps_info, in_air or battery

example adjusted for my hardare port: https://github.com/maciek01/dronegprs-mavsdk/blob/main/mavsdk-examples/telemetry.py

output:

pi@uav7:~/dronegprs-mavsdk/mavsdk-examples $ ./telemetry.py 
Position: [latitude_deg: 42.5224921, longitude_deg: -71.18680859999999, absolute_altitude_m: 55.720001220703125, relative_altitude_m: -0.22700001299381256]
Position: [latitude_deg: 42.522490999999995, longitude_deg: -71.1868108, absolute_altitude_m: 55.67000198364258, relative_altitude_m: -0.23800000548362732]
Position: [latitude_deg: 42.5224901, longitude_deg: -71.1868114, absolute_altitude_m: 55.7400016784668, relative_altitude_m: -0.23800000548362732]
Position: [latitude_deg: 42.522489799999995, longitude_deg: -71.1868094, absolute_altitude_m: 55.85000228881836, relative_altitude_m: -0.20600001513957977]
Position: [latitude_deg: 42.5224869, longitude_deg: -71.1868077, absolute_altitude_m: 56.96000289916992, relative_altitude_m: -0.2110000103712082]
Position: [latitude_deg: 42.522484999999996, longitude_deg: -71.1868093, absolute_altitude_m: 57.2400016784668, relative_altitude_m: -0.2160000056028366]
Position: [latitude_deg: 42.5224833, longitude_deg: -71.1868093, absolute_altitude_m: 58.02000427246094, relative_altitude_m: -0.2110000103712082]
Position: [latitude_deg: 42.5224809, longitude_deg: -71.1868113, absolute_altitude_m: 58.1400032043457, relative_altitude_m: -0.22700001299381256]
Position: [latitude_deg: 42.5225578, longitude_deg: -71.18677869999999, absolute_altitude_m: 3.440000057220459, relative_altitude_m: -0.29200002551078796]
__anext__ grpc exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::1%5D:50051 {created_time:"2023-11-19T18:07:41.912084974-05:00", grpc_status:14, grpc_message:"Socket closed"}"
>
Task exception was never retrieved
future: <Task finished name='Task-5' coro=<print_position() done, defined at /home/pi/dronegprs-mavsdk/mavsdk-examples/./telemetry.py:37> exception=<_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::1%5D:50051 {created_time:"2023-11-19T18:07:41.912084974-05:00", grpc_status:14, grpc_message:"Socket closed"}"
>>
Traceback (most recent call last):
  File "/home/pi/dronegprs-mavsdk/mavsdk-examples/./telemetry.py", line 38, in print_position
    async for position in drone.telemetry.position():
  File "/usr/local/lib/python3.9/dist-packages/mavsdk/telemetry.py", line 3644, in position
    async for response in position_stream:
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 154, in __anext__
    return await asyncio.shield(self._next_future)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "UNKNOWN:Error received from peer ipv6:%5B::1%5D:50051 {created_time:"2023-11-19T18:07:41.912084974-05:00", grpc_status:14, grpc_message:"Socket closed"}"
>
__anext__ grpc exception
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 131, in _next
[... 103 lines cut ...]
	details = "Socket closed"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-11-19T18:07:41.912327981-05:00", grpc_status:14, grpc_message:"Socket closed"}"
>>
Traceback (most recent call last):
  File "/home/pi/dronegprs-mavsdk/mavsdk-examples/./telemetry.py", line 33, in print_in_air
    async for in_air in drone.telemetry.in_air():
  File "/usr/local/lib/python3.9/dist-packages/mavsdk/telemetry.py", line 3692, in in_air
    async for response in in_air_stream:
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 154, in __anext__
    return await asyncio.shield(self._next_future)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 145, in cb
    fut.result()
  File "/usr/lib/python3.9/concurrent/futures/thread.py", line 52, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/local/lib/python3.9/dist-packages/aiogrpc/utils.py", line 131, in _next
    return next(self._iterator)
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 541, in __next__
    return self._next()
  File "/usr/local/lib/python3.9/dist-packages/grpc/_channel.py", line 967, in _next
    raise self
grpc._channel._MultiThreadedRendezvous: <_MultiThreadedRendezvous of RPC that terminated with:
	status = StatusCode.UNAVAILABLE
	details = "Socket closed"
	debug_error_string = "UNKNOWN:Error received from peer  {created_time:"2023-11-19T18:07:41.912327981-05:00", grpc_status:14, grpc_message:"Socket closed"}"
>

JonasVautherin, 2023-11-19:
> 1: The example only reports position telementry. I does not report neither gps_info, in_air or battery

With ardupilot, don't you have to `set_rate` on those telemetry streams? Otherwise I think ardupilot may not be publishing anything by default.

maciek01, 2023-11-19:
let me add it... yes, this helped - thanks!

closed 2023-11-19 by maciek01

maciek01, 2023-11-20:
sorry - i know this has nothign to do with the issue i reported, but since i got your attention :)  - any chance to add the following functionality:

- change desired home position (arducopter style) - important for more precise, GC-assisted RTL
- report current Amp draw (arducopter style)
- report mAh consumed so far (arducopter style)

If these are things related to the actual PX4 i can put my request over there

thanks!

julianoes, 2023-11-20:
>  StatusCode.UNAVAILABLE

Is the mavsdk_server binary crashing?

Can you give me instructions how I can reproduce this?

maciek01, 2023-11-20:
@julianoes no, not since i added the set rate statements. Its been up now for 2 hours and will keep it overnight reporting telemetry. (matek h743 wing, AC 4.2.2, latest mavsdk).