mavlink/MAVSDK-Python #639: goto_location and set_current_speed generate errors both with ArduCopter actual flight and gazebo sim. AC switched to alt_hold when goto requested.

maciek01, 2023-11-24:
gettong the following error in the real flight (AC 4,4,2) as well as Px4 sim gazebo.

Gazebo however respect the specified location and stops the vehicle, while during the real flight, vehocle kept going past the location and was only stopped when i issues RTL.

Attacjed is the screen sht of the AC log analyzer - the selected destination was roughtly in the halfway point belor RTL (geen line) was issued.

Also, not sure why vehicle switched to ALT_HOLD mode when i called goto_location:

```
2023-11-24 08:15:47 INFO     heartbeat sent
2023-11-24 08:15:47 INFO     COMMANDS:{"code":0,"data":{"id":782715,"receivedTimestampMS":1700831747031,"heartbeat":{},"actionRequests":[{"unitId":"uav7","command":{"name":"GOTO","parameters":[{"name":"lat","value":42.52275095338682},{"name":"lon","value":-71.18642723161149}]}}]}}
2023-11-24 08:15:47 INFO     actionRequests: [{"unitId": "uav7", "command": {"name": "GOTO", "parameters": [{"name": "lat", "value": 42.52275095338682}, {"name": "lon", "value": -71.18642723161149}]}}]
2023-11-24 08:15:47 INFO     GOTO
2023-11-24 08:15:47 INFO     MODEM >>
2023-11-24 08:15:47 INFO     MODEM >> +CSQ: 31,99
2023-11-24 08:15:47 INFO     MODEM >> AT+CMGL="ALL"
2023-11-24 08:15:47 INFO     MODEM >> OK
Traceback (most recent call last):
  File "/home/pi/dronegprs-mavsdk/src/pilot.py", line 662, in goto
    await vehicle.action.goto_location(float(requestedLat), float(requestedLon), float(home.absolute_altitude_m) + float(operatingAlt), float(brg))
  File "/usr/local/lib/python3.9/dist-packages/mavsdk/action.py", line 548, in goto_location
    raise ActionError(result, "goto_location()", latitude_deg, longitude_deg, absolute_altitude_m, yaw_deg)
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: goto_location(); params: (42.52275095338682, -71.18642723161149, 167.68000030517578, 119.04294518896484)
Traceback (most recent call last):
  File "/home/pi/dronegprs-mavsdk/src/command_processor.py", line 164, in processCommands
    result = await actions[action['command']['name']](action)
  File "/home/pi/dronegprs-mavsdk/src/command_processor.py", line 59, in goto
    return await pilot.goto(data)
  File "/home/pi/dronegprs-mavsdk/src/pilot.py", line 666, in goto
    await vehicle.action.set_current_speed(float(operatingSpeed))
  File "/usr/local/lib/python3.9/dist-packages/mavsdk/action.py", line 894, in set_current_speed
    raise ActionError(result, "set_current_speed()", speed_m_s)
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: set_current_speed(); params: (15.0,)
2023-11-24 08:15:47 INFO     WAIT FOR ACTION:
2023-11-24 08:15:48 INFO     sending heartbeat
2023-11-24 08:15:48 INFO     heartbeat sent
2023-11-24 08:15:48 INFO     COMMANDS:{"code":0,"data":{"id":782717,"receivedTimestampMS":1700831748212,"heartbeat":{},"actionRequests":null}}
2023-11-24 08:15:49 INFO     sending heartbeat
```
[image]

julianoes, 2023-11-25:
Do you know the mode that ArduCopter wants to be in for goto? I wonder if this change (which is only in main) would work: https://github.com/mavlink/MAVSDK/pull/2170/files

maciek01, 2023-11-25:
the mode was LOITER - the yellow line on the graph. Let me see if i can get this PR assembled locally