mavlink/MAVSDK-Python #640: Connect to QGC while using mavsdk on the companion computer

Hmilzy, 2023-12-05:
Hi all：
I am trying to connect via serial (ttyAMA0) from pixhawk to a raspberryPi and control my drone throught MAVSDK-Python. 
In the meantime, I want to connect my drone to QGC through mavlink-router, but it failed.  
Firmware: PX4 1.13.3
MAVSDK-Python: 1.4.9
mavsdk-server: 1.4.17

First, i run the mavsdk-server manually:  ./mavsdk_server_linux-arm64-musl -p 50051 serial:///dev/ttyAMA0:921600
then, I run the example python code——firmware_version.py  which
    drone = System(mavsdk_server_address='localhost',port=50051)
    await drone.connect(system_address="serial:///dev/ttyAMA0")

[image]
[image]

So I think the above pictures can prove that the connection between the mavsdk and the drone is fine.
Next, I want to connect to QGC on my ubuntu PC 192.168.50.19 by udp, so i run the mavlink-router on raspberryPi like this: 
         mavlink-routerd -e 192.168.50.19:14550 /dev/ttyAMA0:921600   

[image]

At this moment, I found error message in mavsdk_server terminal, and the connection of QGC is very unstable(sometimes ready to fly and sometimes disconnected)
[image]

Finally, I just want to use mavsdk-python at the same time QGC was connected to observe the status of the drone, but the QGC connection remained unstable

What should I do to solve this problem? Please help. Thank you!

Hmilzy, 2023-12-05:
Just to add, The error message of the mavsdk_server termi  is shown in the figure below
[image]

dayjaby, 2023-12-05:
Looks like you are trying to access the same serial device in two places: once with mavsdk-server and once with mavlink-routerd. You can use mavlink-routerd to create a second endpoint:

```
mavlink-routerd -e 192.168.50.19:14550 -e 127.0.0.1:20000 /dev/ttyAMA0:921600
```

and make the mavsdk-server use the new endpoint (assuming that mavlink-routerd and mavsdk-server run on the same machine):

```
./mavsdk_server_linux-arm64-musl -p 50051 udp://0.0.0.0:20000
```

Hmilzy, 2023-12-05:
oh!!!, it's a good idea. Later I will retest according to your guidance.
Thank you very much!

julianoes, 2023-12-05:
@dayjaby answered it perfectly. I'll close this as I think it'll solve the problem.