mavlink/MAVSDK-Python #641: Trying to understand Offboard Position Commands

cameronpm1, 2023-12-05:
I have been trying to control a quadcopter with mavsdk and an onboard flight computer using offboard.set_position_ned, however, my performance is not exactly what I would like, and I am trying to understand why. I will give the drone positional commands of distance about 1-2m from its current position. However, its behavior is not exactly what I would expect. Once offboard mode is engaged I will give it a positional command to move upwards, and then, and then wait several seconds, before giving more positional commands to move forward, however, the drone just continuously loses altitude until it is on the ground. My understanding was when sending a position command to the px4 it will try and maintain this position until a new position is sent. Is this incorrect, and if so how can I get the drone to behave in that manor. And, if I am correct in how the offboard functions should perform, is it likely that my gps is not very acccurate?

here is my code:

print("-- Arming")
        await self.drone.action.arm()
        ned_init = self.get_positionNED()
        ned = ned_init + [0,0,-2,0]
        await self.set_position_NED(NED=ned)

        await self.turn_on_offboard()

        for i in range(10):
            pos = await self.get_position()
            await asyncio.sleep(1)

        await self.set_position_NED(NED=ned+[2,0,0,0])

        for i in range(10):
            pos = await self.get_position()
            await asyncio.sleep(1)

        print("-- Landing")
        await self.land()

        await self.turn_off_offboard()

cameronpm1, 2023-12-06:
Sorry it took a day, is has not stopped snowing by me so I had to wait for a window to fly the drone without worrying about my jetson shorting. 

Here's the link to my log file: https://logs.px4.io/plot_app?log=fd0d176c-f44e-45b5-98db-25447f227ef8

The script I ran is almost identical to the one above, except I give the drone 3 setpoints, 2m up, then 2m up and 1m to the east, and finally 2m up and 2m to the east. It should move in a straight line, but it is not. The logs seem to show the gps data is surprisingly accurate, I just do not understand why the drone is struggling to follow the path I set. Is one possibility that I need to set velocity control as well as positional for it to be more accurate? Thank you!

cameronpm1, 2023-12-14:
After several days of tuning my PID controller the circling has reduced, but the drone still does not hold still in offboard. Is it possible to switch to hold mode in between sending offboard setpoint commands so that the drone stays in a stable position before and after it moves?

julianoes, 2023-12-15:
My guess is that you're experiencing "toilet bowling" because your yaw (as measured by the magnetometer) is off.

cameronpm1, 2023-12-16:
Several times, i've found that after tuning the pid controllers its behavior is much more stable, however, the problem will still randomly arise time to time and then just goes away. Is it possible that my onboard jetson nano is messing with the magnetic field? Or perhaps there is too much vibration? I'm kinda at a loss to why this issue keeps appearing and then randomly going away.

julianoes, 2023-12-16:
One thing to check is to look at the yaw indicated before take off and check whether it is matching the world.

cameronpm1, 2023-12-18:
If it isn't, is there an easy fix or it's an internal problem?

julianoes, 2023-12-19:
It sounds like a PX4 issue that you need to get to the bottom of. Not sure MAVSDK can do much about it.