mavlink/MAVSDK-Python #642: Timeout error trying to use gimbal, PX4

whitney-smith, 2023-12-05:
I'm getting a timeout error while trying to run gimbal.py from the examples. I can control the gimbal with QGC so the gimbal is exposed over mavlink. PX4 version is v1.12.

```
Traceback (most recent call last):
  File "/home/whitney/Documents/Dev/Robot/./gimbal.py", line 89, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 646, in run_until_complete
    return future.result()
  File "/home/whitney/Documents/Dev/./gimbal.py", line 25, in run
    await drone.gimbal.set_mode(GimbalMode.YAW_FOLLOW)
  File "/home/whitney/.local/lib/python3.10/site-packages/mavsdk/gimbal.py", line 484, in set_mode
    raise GimbalError(result, "set_mode()", gimbal_mode)
mavsdk.gimbal.GimbalError: TIMEOUT: 'Timeout'; origin: set_mode(); params: (<GimbalMode.YAW_FOLLOW: 0>,)
```

julianoes, 2023-12-06:
PX4 v1.12 likely does not support gimbal v2 fully yet, so I'm not sure if it'll work out of the box.

Have you set this param to gimbal v2?
https://docs.px4.io/v1.12/en/advanced_config/parameter_reference.html#MNT_MODE_IN

whitney-smith, 2023-12-08:
I'm using gimbal protocol v1. I set MNT_MODE_IN to 3 but same result unfortunately

julianoes, 2023-12-10:
Any chance you could try with a more recent PX4 version? I'm not sure of the state of gimbal back then.

whitney-smith, 2024-03-20:
I found the solution. I needed to set MNT_MODE_OUT in addition to MNT_MODE_IN
MNT_MODE_IN 3
MNT_MODE_OUT 1

closed 2024-03-21 by julianoes