mavlink/MAVSDK-Python #643: the script didnt work

naorwaiss, 2023-12-08:
i try to run this script at the simulation and the script run and do what the drone need to do 
after that i try it with real drone with gps 13 satellite and hdop<0.9 and the drone didnt move 

###

# this simulation is only move the drone and gave him command this code also reset the position of the drone to new orgin)

import asyncio
from mavsdk import System
from mavsdk.offboard import (OffboardError, PositionNedYaw)

MOVE_ACCURACY = 0.1

async def setup_drone(drone: System):
    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()
    await drone.action.takeoff()

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))
    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed \
                with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        raise

async def land(drone: System):
    await move(drone, 0.0, 0.0, 0.0)
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed \
                with error code: {error._result.result}")
    await drone.action.land()

async def move(drone: System, north_m: float, east_m: float, down_m: float, yaw_deg: float = 0.0):
    await drone.offboard.set_position_ned(
        PositionNedYaw(north_m, east_m, down_m, yaw_deg))
    while True:
        await asyncio.sleep(0.1)
        position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
        north_m_distance = position_ned.position.north_m - north_m
        east_m_distance = position_ned.position.east_m - east_m
        down_m_distance = position_ned.position.down_m - down_m
        if (
                MOVE_ACCURACY > north_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > east_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > down_m_distance > -MOVE_ACCURACY
        ):
            break

async def move_loop(drone: System):
    position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
    position = position_ned.position
    print(f">> Current position ({position.north_m}, {-position.east_m}, {-position.down_m})")
    direction = input("Enter direction [forward, right, up, land]: ")
    if direction == "land":
        await land(drone)
    else:
        distance = float(input("Enter distance [in meters]: "))
        target_north_m = position.north_m
        target_east_m = position.east_m
        target_down_m = position.down_m
        if direction == "forward":
            target_north_m += distance
        elif direction == "right":
            target_east_m += distance
        elif direction == "up":
            target_down_m -= distance
        await move(drone, target_north_m, target_east_m, target_down_m)

async def main():
    drone = System()
    # await drone.connect(system_address="udp://:14540")
    await drone.connect(system_address="serial:///dev/ttyTHS1")
    await setup_drone(drone)

    height = float(input(
        "Enter drone height [in meters]: "
    ))

    async for position_ned in drone.telemetry.position_velocity_ned():
        position = position_ned.position
        await move(drone, position.north_m, position.east_m, -height)
        break

    while True:
        try:
            await move_loop(drone)
        except Exception as e:
            print(f">> Failed with {e}, landing")
            await land(drone)

if __name__ == "__main__":
    asyncio.run(main())
###

thanks for the help

JonasVautherin, 2023-12-08:
I am afraid you will have to do some debugging here. For instance, did the script detect the drone? Did the drone arm? If not, what happened?

naorwaiss, 2023-12-09:
the script detect the drone the drone arm 
then the drone didn't open the throttle to left off
thanks

JonasVautherin, 2023-12-09:
I'm sorry but your code above is barely readable the way it is formatted, and it is hard for me to just guess what could have possibly happened after it armed :thinking: :confused:.

naorwaiss, 2023-12-13:
onother question/// i try to make script that control the drone with the gps 
right now i try to move from geo coordinate to cartazian cordinate 
i have here some problem with the calculation can help me ?

"""
import asyncio
from mavsdk import System
import math
global x_initial , y_initial , z_initial,  latitude_i ,longitude_i ,altitude_i
x_initial = y_initial = z_initial =latitude_i = longitude_i = altitude_i= 0

async def get_location(drone):
    # this function only take the position from the drone at geo coordinate
    default_coordinates = (0, 0, 0)  # Provide default coordinates if telemetry data is not available

    try:
        async for position in drone.telemetry.position():
            latitude = position.latitude_deg
            longitude = position.longitude_deg
            altitude = position.absolute_altitude_m
            break  # Stop after obtaining coordinates once
    except Exception as e:
        print(f"Error getting location: {e}")
        print("Using default coordinates.")
        return default_coordinates

    return latitude, longitude, altitude

async def convert_geodetic_to_cartesian(latitude, longitude, altitude):
    # Constants
    a = 6378137.0  # Semi-major axis of the Earth in meters
    f = 1 / 298.257223563  # Flattening
    e_sq = f * (2 - f)  # Eccentricity squared

    # Convert latitude and longitude to radians
    lat_rad = math.radians(latitude)
    lon_rad = math.radians(longitude)

    # Calculate N value
    N = a / math.sqrt(1 - e_sq * math.sin(lat_rad)**2)
    # Calculate Cartesian coordinates
    x = (N + altitude) * math.cos(lat_rad) * math.cos(lon_rad)
    y = (N + altitude) * math.cos(lat_rad) * math.sin(lon_rad)
    z = (N * (1 - e_sq) + altitude) * math.sin(lat_rad)

    return x, y, z

async def position_new_cartzian(drone):
    latitude = longitude = altitude = y_n = z_n = x_rel = y_rel = z_rel = 0
    try:
        latitude, longitude, altitude = await get_location(drone)
        x_n, y_n, z_n = await convert_geodetic_to_cartesian(latitude, longitude, altitude)

        x_rel = (x_n - x_initial)
        y_rel = (y_n - y_initial)
        z_rel = (z_n - z_initial)

        return x_rel, y_rel, z_rel
    except Exception as e:
        print(f"Error getting new position: {e}")
    return 0, 0, 0  # Return zeros if there is an error

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f"Status: {status_text.type}: {status_text.text}")
    except asyncio.CancelledError:
        return

async def run():

    drone = System()
    await drone.connect(system_address="udp://:14540")

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    try:
        print("Waiting for drone to connect...")
        async for state in drone.core.connection_state():
            if state.is_connected:
                print(f"-- Connected to drone!")
                break

        print("Waiting for drone to have a global position estimate...")
        async for health in drone.telemetry.health():
            if health.is_global_position_ok and health.is_home_position_ok:
                print("-- Global position estimate OK")
                break

        # check the gps functionalety  , print the two type of gps - olmost onley to the debug
        latitude_i, longitude_i, altitude_i = await get_location(drone)
        print(f"At GPS position: Latitude={latitude_i}, Longitude={longitude_i}, Altitude={altitude_i} meters")
        x_initial, y_initial, z_initial =await convert_geodetic_to_cartesian(longitude_i, latitude_i, altitude_i)
        print(f"At Cartesian position: x={x_initial}, y={y_initial}, z={z_initial}")
        print(await position_new_cartzian(drone))

        print("-- Arming")
        await drone.action.arm()

        print("-- Taking off")
        target_altitude = int(input("Enter the target altitude in meters: "))
        await drone.action.set_takeoff_altitude(target_altitude)
        await drone.action.takeoff()

        await asyncio.sleep(15)
        print(await get_location(drone))
        print("-- Landing")
        await drone.action.land()

    finally:
        # Properly cancel and await completion of status_text_task
        status_text_task.cancel()
        await asyncio.gather(status_text_task, return_exceptions=True)

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
    """
    
    the output 
    """
    aiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
At GPS position: Latitude=47.397749399999995, Longitude=8.5455994, Altitude=488.1180114746094 meters
At Cartesian position: x=4270102.3541171625, y=4643335.343387359, z=941566.116468123
(4277550.588335014, 642765.001239949, 4672169.329376612)
-- Arming
"""
at the line befor the arming you can see that the calclate didnt work
thanks for the help

julianoes, 2023-12-15:
Please don't hijack the thread for new questions.

closed 2023-12-15 by julianoes