mavlink/MAVSDK-Python #644: Qgroundcontrol import function it's _**NOT** currently working_ as expected

igoutta, 2023-12-19:
Hi, I am wondering if you are aware that the import function it's _**NOT** currently working_ as excepted. 

I'm currently developing a computer companion module with your SDK and the problem is the following. The system is commanded to load a mission previously acquired through QGroundControl. The mission in question it's really simple, just a few points to go through. The problem arises at the moment the system is commanded to upload the mission, because the error shows the invalid argument description. I researched in this repo the details of this issue and the I encountered that such a problem usually happens when the **sequence is bad** and/or the current number is more than one in the entire mission. Then I printed the items processed by the SDK and then I found a double zero at the first two items, differents frames that really shouldn't be a problem, current value in the second item, and the param 4 is zero at the first item. The first item apparently is taken from the planned home position of the Qgroundcontrol plan which only has 3 values corresponded to the drone's 3D position. So I'm currently bypassing that first item but I'm looking forward to a solution in this matter. 

[image]

Qgroundcontrol mission plan: 
```
{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 1,
        "firmwareType": 3,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    0,
                    -2.1157681120467093,
                    -79.9385202714547,
                    0
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    -2.1158699615280367,
                    -79.93851893061948,
                    0
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 0,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    0,
                    0,
                    0,
                    null,
                    -2.116022744157194,
                    -79.93852294714648,
                    0
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            -2.295260803823127,
            -79.7604224,
            0
        ],
        "vehicleType": 10,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}

```
Thanks for reading and sorry for the trouble.

_Originally posted by @igoutta in https://github.com/mavlink/MAVSDK-Python/issues/439#issuecomment-1863451170_

dayjaby, 2023-12-19:
What kind of autopilot are you trying to upload the mission to?

https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/mission_raw/mission_import.cpp#L118 does the seq numbering, which looks correct. But after that, there is one exception: if it's ardupilot, it adds another mission item with seq 0 to the start of the list. Moving the sequencing to after this ardupilot home waypoint insertion should solve the issue.

Do you know how to compile MAVSDK to try it yourself? :)

igoutta, 2023-12-20:
Yes, indeed. I'm using Ardupilot firmware, Boat framing in Ardurover selection if you want even more details. 

When I opened the issue I had already implemented the command to pop that first item.

For the time, I'm going to stick to the compiled version of MAVSDK.

I wish the documentation could be a bit cleared about this tricks. Do you know which way I can contribute there?

igoutta, 2023-12-20:
Yes, in fact I am using the Ardupilot firmware, Boat framing in Ardurover selection if you want more details. 

When I opened the topic I had already implemented the command to pop that first element. And I'm going to stick to the compiled version of MAVSDK for the moment.

I wish the documentation could be a bit cleared about this tricks. Do you know which way I can contribute there?

closed 2023-12-20 by igoutta

dayjaby, 2023-12-20:
Try the compilation guideline in https://mavsdk.mavlink.io/main/en/cpp/guide/build.html up to the `sudo cmake --build build/default --target install` step.

Well, it's a bug that needs to be fixed. Documentation won't really help with that. If you change the MAVSDK code and find a working fix, feel free to contribute by opening a pull request.

julianoes, 2023-12-20:
FYI v2 will be out soon.

igoutta, 2023-12-30:
Really? Wish to see that changelog.

julianoes, 2024-01-02:
@igoutta https://github.com/mavlink/MAVSDK/releases/tag/v2.0.0