mavlink/MAVSDK-Python #645: global possition asstimate

naorwaiss, 2023-12-21:
hi, try to gave my script with ned (the code in the end ) and the drone say that the drone wait to global position estimate.
the code stuck at this position. can i get explain how to push the drone to get this value 
thanks

julianoes, 2023-12-23:
Which PX4 version, or ArduPilot version is this? How is it connected? Does it have GPS? What's the console output?

Have you tried debugging using these instructions: http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/#debug-connection-issues

naorwaiss, 2023-12-23:
hi 
px4 version 1.14v
yes the drone have GPS (11+ sat ,hdop <=1) 
the connect is via the rx tx and ground - the connection is work well and i check it with mavlink 
what is this console, sry i didnt know how to updn console at px4 but the drone is armable and i can do takeoff and land with the QGC and mavlink 
thanks

julianoes, 2023-12-23:
You could try to arm and takeoff anyway, without waiting for that flag. I'm not sure why it doesn't pass. You would have to share some log output, screenshots, video, logs, anything.

naorwaiss, 2023-12-24:
'''
# this simulation is only move the drone and gave him command this code also reset the position of the drone to new orgin)

import asyncio
from mavsdk import System
from mavsdk.offboard import (OffboardError, PositionNedYaw)

MOVE_ACCURACY = 0.1

async def setup_drone(drone: System):
    print("-- Arming")
    await drone.action.arm()
    await drone.action.takeoff()

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))
    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed \
                with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        raise

async def land(drone: System):
    await move(drone, 0.0, 0.0, 0.0)
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed \
                with error code: {error._result.result}")
    await drone.action.land()

async def move(drone: System, north_m: float, east_m: float, down_m: float, yaw_deg: float = 0.0):
    await drone.offboard.set_position_ned(
        PositionNedYaw(north_m, east_m, down_m, yaw_deg))
    while True:
        await asyncio.sleep(0.1)
        position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
        north_m_distance = position_ned.position.north_m - north_m
        east_m_distance = position_ned.position.east_m - east_m
        down_m_distance = position_ned.position.down_m - down_m
        if (
                MOVE_ACCURACY > north_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > east_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > down_m_distance > -MOVE_ACCURACY
        ):
            break

async def move_loop(drone: System):
    position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
    position = position_ned.position
    print(f">> Current position ({position.north_m}, {-position.east_m}, {-position.down_m})")
    direction = input("Enter direction [forward, right, up, land]: ")
    if direction == "land":
        await land(drone)
    else:
        distance = float(input("Enter distance [in meters]: "))
        target_north_m = position.north_m
        target_east_m = position.east_m
        target_down_m = position.down_m
        if direction == "forward":
            target_north_m += distance
        elif direction == "right":
            target_east_m += distance
        elif direction == "up":
            target_down_m -= distance
        await move(drone, target_north_m, target_east_m, target_down_m)

async def main():
    drone = System()
    await drone.connect(system_address="udp://:14540")
    #await drone.connect(system_address="serial:///dev/ttyTHS1")
    await setup_drone(drone)

    height = float(input(
        "Enter drone height [in meters]: "
    ))

    async for position_ned in drone.telemetry.position_velocity_ned():
        position = position_ned.position
        await move(drone, position.north_m, position.east_m, -height)
        break

    while True:
        try:
            await move_loop(drone)
        except Exception as e:
            print(f">> Failed with {e}, landing")
            await land(drone)

if __name__ == "__main__":
    asyncio.run(main())
    '''
    at this code i try to remove the flage -- this is the debug from the console 
    '''
    [03:42:44|Debug] MAVLink: critical: Switching to AUTO_TAKEOFF is currently not available	 (system_impl.cpp:250)
[03:42:52|Debug] MAVLink: info: Disarmed by auto preflight disarming (system_impl.cpp:250)
[03:43:01|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[03:43:01|Warn ] command temporarily rejected (22). (mavlink_command_sender.cpp:205)
[03:43:02|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/13_43_01.ulg	 (system_impl.cpp:250)
[03:43:03|Debug] MAVLink: critical: Switching to AUTO_TAKEOFF is currently not available	 (system_impl.cpp:250)
[03:43:04|Debug] MAVLink: critical: Switching to OFFBOARD is currently not available	 (system_impl.cpp:250)
[03:43:05|Debug] MAVLink: critical: Failsafe activated	 (system_impl.cpp:250)
[03:43:06|Debug] MAVLink: info: Disarmed by RC	 (system_impl.cpp:250)

'''

naorwaiss, 2023-12-24:
i add here the privious code with de debug (wait to global positon)
```
# this simulation is only move the drone and gave him command this code also reset the position of the drone to new orgin)

import asyncio
from mavsdk import System
from mavsdk.offboard import (OffboardError, PositionNedYaw)

MOVE_ACCURACY = 0.1

async def setup_drone(drone: System):
    print("Waiting for drone to have a global position estimate...")
    async for health in drone.telemetry.health():
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position estimate OK")
            break

    print("-- Arming")
    await drone.action.arm()
    await drone.action.takeoff()

    print("-- Setting initial setpoint")
    await drone.offboard.set_position_ned(PositionNedYaw(0.0, 0.0, 0.0, 0.0))
    print("-- Starting offboard")
    try:
        await drone.offboard.start()
    except OffboardError as error:
        print(f"Starting offboard mode failed \
                with error code: {error._result.result}")
        print("-- Disarming")
        await drone.action.disarm()
        raise

async def land(drone: System):
    await move(drone, 0.0, 0.0, 0.0)
    try:
        await drone.offboard.stop()
    except OffboardError as error:
        print(f"Stopping offboard mode failed \
                with error code: {error._result.result}")
    await drone.action.land()

async def move(drone: System, north_m: float, east_m: float, down_m: float, yaw_deg: float = 0.0):
    await drone.offboard.set_position_ned(
        PositionNedYaw(north_m, east_m, down_m, yaw_deg))
    while True:
        await asyncio.sleep(0.1)
        position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
        north_m_distance = position_ned.position.north_m - north_m
        east_m_distance = position_ned.position.east_m - east_m
        down_m_distance = position_ned.position.down_m - down_m
        if (
                MOVE_ACCURACY > north_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > east_m_distance > -MOVE_ACCURACY
                and MOVE_ACCURACY > down_m_distance > -MOVE_ACCURACY
        ):
            break

async def move_loop(drone: System):
    position_ned = await drone.telemetry.position_velocity_ned().__aiter__().__anext__()
    position = position_ned.position
    print(f">> Current position ({position.north_m}, {-position.east_m}, {-position.down_m})")
    direction = input("Enter direction [forward, right, up, land]: ")
    if direction == "land":
        await land(drone)
    else:
        distance = float(input("Enter distance [in meters]: "))
        target_north_m = position.north_m
        target_east_m = position.east_m
        target_down_m = position.down_m
        if direction == "forward":
            target_north_m += distance
        elif direction == "right":
            target_east_m += distance
        elif direction == "up":
            target_down_m -= distance
        await move(drone, target_north_m, target_east_m, target_down_m)

async def main():
    drone = System()
    # await drone.connect(system_address="udp://:14540")
    await drone.connect(system_address="serial:///dev/ttyTHS1")
    await setup_drone(drone)

    height = float(input(
        "Enter drone height [in meters]: "
    ))

    async for position_ned in drone.telemetry.position_velocity_ned():
        position = position_ned.position
        await move(drone, position.north_m, position.east_m, -height)
        break

    while True:
        try:
            await move_loop(drone)
        except Exception as e:
            print(f">> Failed with {e}, landing")
            await land(drone)

if __name__ == "__main__":
    asyncio.run(main())

```

```

[03:49:09|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[03:49:09|Info ] Waiting to discover system on serial:///dev/ttyTHS1... (connection_initiator.h:20)
[03:49:09|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[03:49:09|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[03:49:09|Debug] New: System ID: 51 Comp ID: 68 (mavsdk_impl.cpp:496)
[03:49:09|Debug] Component Unsupported component (68) added. (system_impl.cpp:377)
[03:49:10|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[03:49:10|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[03:49:10|Info ] System discovered (connection_initiator.h:63)
[03:49:10|Info ] Server started (grpc_server.cpp:53)
[03:49:10|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[03:49:10|Warn ] sending again after 0.501207 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[03:49:30|Warn ] sending again, retries to do: 3  (CAL_GYRO0_ID). (mavlink_parameters.cpp:1291)
[03:49:31|Warn ] sending again, retries to do: 2  (CAL_GYRO0_ID). (mavlink_parameters.cpp:1291)
[03:49:31|Warn ] sending again, retries to do: 3  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[03:49:32|Warn ] sending again, retries to do: 2  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[03:49:32|Warn ] sending again, retries to do: 3  (SYS_HITL). (mavlink_parameters.cpp:1291)
[03:49:33|Warn ] sending again, retries to do: 2  (SYS_HITL). (mavlink_parameters.cpp:1291)
[03:49:33|Warn ] sending again, retries to do: 1  (SYS_HITL). (mavlink_parameters.cpp:1291)
[03:49:34|Error] Error: Retrying failed get param busy timeout: SYS_HITL (mavlink_parameters.cpp:1320)
[03:49:34|Error] Error: Param to determine hitl failed. (telemetry_impl.cpp:1691)
[03:49:35|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:49:37|Info ] heartbeats timed out (system_impl.cpp:272)
[03:49:39|Debug] Discovered 1 component(s) (system_impl.cpp:578)

```

naorwaiss, 2023-12-24:
```
[03:59:01|Info ] heartbeats timed out (system_impl.cpp:272)
[03:59:02|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[03:59:02|Warn ] sending again after 0.505617 s, retries to do: 3  (511). (mavlink_command_sender.cpp:287)
[03:59:02|Warn ] sending again after 0.506855 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[03:59:08|Debug] MAVLink: info: Disarmed by auto preflight disarming	 (system_impl.cpp:250)
[03:59:20|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[03:59:21|Warn ] sending again after 0.502296 s, retries to do: 3  (22). (mavlink_command_sender.cpp:287)
[03:59:21|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:21|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/13_59_20.ulg	 (system_impl.cpp:250)
[03:59:21|Warn ] sending again after 1.00608 s, retries to do: 2  (22). (mavlink_command_sender.cpp:287)
[03:59:22|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:22|Warn ] sending again after 0.502714 s, retries to do: 3  (176). (mavlink_command_sender.cpp:287)
[03:59:22|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:22|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (system_impl.cpp:250)
[03:59:29|Info ] heartbeats timed out (system_impl.cpp:272)
[03:59:33|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[03:59:35|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[03:59:36|Warn ] sending again after 0.502648 s, retries to do: 3  (176). (mavlink_command_sender.cpp:287)
[03:59:36|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/13_59_35.ulg	 (system_impl.cpp:250)
[03:59:37|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (system_impl.cpp:250)
[03:59:38|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:42|Info ] heartbeats timed out (system_impl.cpp:272)
[03:59:45|Debug] MAVLink: info: Disarmed by auto preflight disarming	 (system_impl.cpp:250)
[03:59:47|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[03:59:52|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[03:59:53|Warn ] sending again after 0.502416 s, retries to do: 3  (22). (mavlink_command_sender.cpp:287)
[03:59:53|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:53|Warn ] sending again after 0.505004 s, retries to do: 3  (176). (mavlink_command_sender.cpp:287)
[03:59:55|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m (system_impl.cpp:250)
[03:59:56|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:57|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:57|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[03:59:58|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:00:00|Info ] heartbeats timed out (system_impl.cpp:272)
[04:00:00|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:00:00|Warn ] sending again after 0.50272 s, retries to do: 3  (511). (mavlink_command_sender.cpp:287)
[04:00:00|Warn ] sending again after 0.503908 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[04:00:01|Warn ] sending again after 1.0121 s, retries to do: 2  (520). (mavlink_command_sender.cpp:287)
[04:00:01|Warn ] sending again after 1.01309 s, retries to do: 2  (511). (mavlink_command_sender.cpp:287)
[04:00:02|Debug] MAVLink: info: Disarmed by auto preflight disarming	 (system_impl.cpp:250)
[04:00:04|Info ] heartbeats timed out (system_impl.cpp:272)
[04:00:05|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:00:24|Debug] MAVLink: info: Armed by external command	 (system_impl.cpp:250)
[04:00:25|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/14_00_24.ulg	 (system_impl.cpp:250)
[04:00:26|Debug] MAVLink: info: Takeoff detected	 (system_impl.cpp:250)
[04:00:33|Debug] MAVLink: info: Pilot took over using sticks	 (system_impl.cpp:250)
[04:00:40|Debug] MAVLink: critical: Switching to OFFBOARD is currently not available	 (system_impl.cpp:250)
[04:00:46|Debug] MAVLink: info: Landing detected	 (system_impl.cpp:250)
[04:00:47|Debug] MAVLink: info: Disarmed by RC	 (system_impl.cpp:250)
[04:01:28|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:01:30|Info ] heartbeats timed out (system_impl.cpp:272)
[04:01:30|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:01:30|Warn ] sending again after 0.502834 s, retries to do: 3  (511). (mavlink_command_sender.cpp:287)
[04:01:30|Warn ] sending again after 0.503956 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)
[04:01:30|Warn ] sending again after 0.504652 s, retries to do: 3  (400). (mavlink_command_sender.cpp:287)
[04:01:31|Warn ] sending again after 1.01207 s, retries to do: 2  (400). (mavlink_command_sender.cpp:287)
[04:01:31|Warn ] sending again after 1.0128 s, retries to do: 2  (520). (mavlink_command_sender.cpp:287)
[04:01:31|Warn ] sending again after 1.01319 s, retries to do: 2  (511). (mavlink_command_sender.cpp:287)
[04:01:31|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/14_01_30.ulg	 (system_impl.cpp:250)
[... cut ...]
[04:01:36|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:01:37|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:01:40|Info ] heartbeats timed out (system_impl.cpp:272)
[04:01:41|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:01:41|Warn ] sending again after 0.501401 s, retries to do: 3  (520). (mavlink_command_sender.cpp:287)

```
this is the console after i try to run the code few time with the flage

naorwaiss, 2023-12-24:
```
p:578)
^C
drone@drone-desktop:~$ /home/drone/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyTHS1
[04:26:43|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[04:26:43|Info ] Waiting to discover system on serial:///dev/ttyTHS1... (connection_initiator.h:20)
[04:26:43|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[04:26:43|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[04:26:43|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[04:26:43|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:26:43|Info ] System discovered (connection_initiator.h:63)
[04:26:43|Info ] Server started (grpc_server.cpp:53)
[04:26:43|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[04:26:43|Debug] New: System ID: 51 Comp ID: 68 (mavsdk_impl.cpp:496)
[04:26:43|Debug] Component Unsupported component (68) added. (system_impl.cpp:377)
^C
drone@drone-desktop:~$ clear

drone@drone-desktop:~$ /home/drone/.local/lib/python3.9/site-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyTHS1
[04:27:04|Info ] MAVSDK version: v1.4.16 (mavsdk_impl.cpp:20)
[04:27:04|Info ] Waiting to discover system on serial:///dev/ttyTHS1... (connection_initiator.h:20)
[04:27:04|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[04:27:04|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[04:27:04|Debug] MAVLink: info: Landing detected	 (system_impl.cpp:250)
[04:27:04|Debug] New: System ID: 51 Comp ID: 68 (mavsdk_impl.cpp:496)
[04:27:04|Debug] Component Unsupported component (68) added. (system_impl.cpp:377)
[04:27:04|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:225)
[04:27:04|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:27:04|Info ] System discovered (connection_initiator.h:63)
[04:27:04|Info ] Server started (grpc_server.cpp:53)
[04:27:04|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[04:27:05|Debug] MAVLink: info: Disarmed by RC	 (system_impl.cpp:250)
[04:27:09|Warn ] sending again, retries to do: 3  (CAL_ACC0_ID). (mavlink_parameters.cpp:1291)
[04:27:09|Warn ] sending again, retries to do: 3  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[04:27:10|Warn ] sending again, retries to do: 2  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[04:27:10|Warn ] sending again, retries to do: 1  (CAL_MAG0_ID). (mavlink_parameters.cpp:1291)
[04:27:11|Error] Error: Retrying failed get param busy timeout: CAL_MAG0_ID (mavlink_parameters.cpp:1320)
[04:27:11|Error] Error: Param for mag cal failed. (telemetry_impl.cpp:1554)
[04:27:11|Info ] heartbeats timed out (system_impl.cpp:272)
[04:27:11|Warn ] sending again, retries to do: 3  (SYS_HITL). (mavlink_parameters.cpp:1291)
[04:27:16|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:27:16|Debug] MAVLink: info: Armed by external command (system_impl.cpp:250)
[04:27:17|Warn ] sending again after 0.501151 s, retries to do: 3  (22). (mavlink_command_sender.cpp:287)
[04:27:17|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/14_27_16.ulg	 (system_impl.cpp:250)
[04:27:17|Warn ] sending again after 1.00985 s, retries to do: 2  (22). (mavlink_command_sender.cpp:287)
[04:27:18|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m	 (system_impl.cpp:250)
[04:27:19|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:27:19|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m	 (system_impl.cpp:250)
[04:27:20|Debug] MAVLink: info: Takeoff detected	 (system_impl.cpp:250)
[04:27:21|Debug] MAVLink: info: Using minimum takeoff altitude: 2.50 m	 (system_impl.cpp:250)
[04:27:22|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:27:24|Info ] heartbeats timed out (system_impl.cpp:272)
[04:27:24|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:27:27|Debug] MAVLink: info: Armed by RC	 (system_impl.cpp:250)
[04:27:28|Debug] MAVLink: info: [logger] /fs/microsd/log/2023-12-24/14_27_26.ulg	 (system_impl.cpp:250)
[04:27:28|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
[04:27:29|Debug] MAVLink: info: Takeoff detected	 (system_impl.cpp:250)
[04:27:30|Info ] heartbeats timed out (system_impl.cpp:272)
[04:27:30|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[04:27:32|Warn ] Received ack for not-existing command: 520! Ignoring... (mavlink_command_sender.cpp:258)
^C
drone@drone-desktop:~$ 
```
at this time the drone accept the command but it drift to the left at take off (hard left)

naorwaiss, 2023-12-27:
i dont have a way to get this logs... i try another way to resolve this problem base on go to 
i make new issue about it 
thanks alot

julianoes, 2023-12-27:
Ok, that's odd that you can't get logs. Hard to get stuff done without it.

closed 2023-12-27 by julianoes