mavlink/MAVSDK-Python #646: Connection issue

DevankG, 2023-12-27:
Hi
I am using an orange cubepilot with a siyi A8 gimbal attached to it.
I am able to move the gimbal from mission planner and now want to control it from mavsdk-python.
I installed the mavsdk using pip install command got the following:
This installation does not provide an embedded 'mavsdk_server' binary.
If you installed using pip, this means that 'mavsdk_server' is not distributed
for your platform yet (e.g. arm).

You will need to get and run the 'mavsdk_server' binary manually:

  1. Download 'mavsdk_server' from: https://github.com/mavlink/mavsdk/releases
     or build it from source.

  2. Run it, e.g. on port 50051:
     './mavsdk_server -p 50051'

  3. Set the 'mavsdk_server_address' and port when creating the System:
     'drone = System(mavsdk_server_address='localhost', port=50051)'
     
    I am using a windows 11 laptop for this
    I have browsed through many similar issues but could not find a working solution.
    I have also tried downloading the mavsdk server from releases but still nothings works instead of the above mentioned instructions, I did not get anything, program just simply runs and nothing comes in terminal

JonasVautherin, 2023-12-27:
Did you try following the steps in the error message? What is unclear for you there?

julianoes, 2023-12-27:
Is your laptop arm or x86_64? We don't have a Windows arm build yet.

DevankG, 2023-12-28:
> Did you try following the steps in the error message? What is unclear for you there?

Yes, I did. I followed the instructions and nothing happened. The program did not give any errors but the terminal remained blank.

DevankG, 2023-12-28:
> Is your laptop arm or x86_64? We don't have a Windows arm build yet.

X86_64

JonasVautherin, 2023-12-28:
Which `mavsdk_server` executable did you run, and what was the output of that?