mavlink/MAVSDK-Python #648: Error when using MAV_CMD_CONDITION_YAW with MissionRaw plugin.

mfattori, 2024-01-08:
Hi, 

I am using the MissionRaw plugin and am attempting to add a command which manually sets the Yaw angle of the drone. Following the command structure [here](https://mavlink.io/en/messages/common.html#MAV_CMD_CONDITION_YAW), gives the following error: 
```py
TypeError: None has type NoneType, but expected one of: int
```
since MissionRaw expects `x`/`y`/`z` to all be ints. 

If I change them to all 0s, I get a different error:

```py
mavsdk.mission_raw.MissionRawError: UNSUPPORTED: 'Unsupported'; origin: upload_mission(); params: ([<mavsdk.mission_raw.MissionItem object at 0x7ff0b0756ee0>, <mavsdk.mission_raw.MissionItem object at 0x7ff0b0756f40>, <mavsdk.mission_raw.MissionItem object at 0x7ff0b0756a90>],)
``` 

## Minimal example:

Based off the [Mission Raw example](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/mission_raw.py) and adding in a command to change the yaw. 

```py
import asyncio
from mavsdk import System
from mavsdk import mission_raw

async def px4_connect_drone():
    drone = System()
    await drone.connect(system_address="udp://:14540")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            return drone

async def run():
    drone = await px4_connect_drone()
    await run_drone(drone)

async def run_drone(drone):

    mission_items = []

    mission_items.append(mission_raw.MissionItem(
         0,  # start seq at 0
         6,
         16,
         1,  # first one is current
         1,
         0, 10, 0, float('nan'),
         int(47.40271757 * 10**7),
         int(8.54285027 * 10**7),
         30.0,
         0
     ))

    mission_items.append(mission_raw.MissionItem(
        1,
        6,
        16,
        0,
        1,
        0, 10, 0, float('nan'),
        int(47.40271757 * 10**7),
        int(8.54361892 * 10**7),
        30.0,
        0
    ),
    )

    # NEW: Change yaw following 
    # https://mavlink.io/en/messages/common.html#MAV_CMD_CONDITION_YAW
    mission_items.append(mission_raw.MissionItem(
        2,
        6,
        115,
        0,
        1,
        param1 = 340.0,
        param2 = 1.0,
        param3 = 0.0,
        param4 = 0.0, 
        x = 0, y = 0, z = 0.0, # changing x/y/z to None also doesn't work
        mission_type=0
    ),
    )

    print("-- Uploading mission")
    await drone.mission_raw.upload_mission(mission_items)
    print("-- Done")

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())
```

## Versions
Autopilot: PX4/SITL
MAVSDK Python: mavsdk==1.4.9
mavlink-router: v3-11-g42529d5

JonasVautherin, 2024-01-08:
It doesn't want `None` for x/y/z because those should be int, I guess: https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission_raw/mission_raw.proto#L202-L204.

Then it feels like the `UNSUPPORTED` happens in `upload_mission`, but I am not sure why :thinking:. Are you sure that the named arguments work(`param1 = ...`, `mission_type = ...`)? Your other mission items don't have them.

mfattori, 2024-01-08:
I don't think named arguments are the issue. I have some internal code using MissionRaw for waypoints with named arguments and everything works smoothly. 

As a check, I tried removing the named arguments in the above example to match the other items:
```py
...
mission_items.append(mission_raw.MissionItem(
        2,
        6,
        115,
        0,
        1,
        340.0,
        1.0,
        0.0,
        0.0, 
        0,0,0.0,
        0
    ),
    )
...
```
and them same error occurs.

JonasVautherin, 2024-01-08:
And does it work without that specific `MissionItem`? Or does it fail everytime you call `upload_mission` with mission items?

mfattori, 2024-01-08:
The script runs successfully without that specific MissionItem.

JonasVautherin, 2024-01-08:
Could it be that the autopilot says that the mission is unsupported? https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/mission_raw/mission_raw.proto#L226

Did you try uploading the same mission from e.g. QGC?

mfattori, 2024-01-08:
Attempting to upload the same mission from QGC gives the following error: 

"Mission transfer failed. Error: Command is not supported. Item 2 Command: Wait for Yaw".

The plan file is: 
```json
{
    "fileType": "Plan",
    "geoFence": {
        "circles": [
        ],
        "polygons": [
        ],
        "version": 2
    },
    "groundStation": "QGroundControl",
    "mission": {
        "cruiseSpeed": 15,
        "firmwareType": 12,
        "globalPlanAltitudeMode": 1,
        "hoverSpeed": 5,
        "items": [
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 30,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 1,
                "frame": 3,
                "params": [
                    0,
                    10,
                    0,
                    null,
                    47.4027175,
                    8.5428502,
                    30
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 30,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 16,
                "doJumpId": 2,
                "frame": 3,
                "params": [
                    0,
                    10,
                    0,
                    null,
                    47.4027175,
                    8.5436189,
                    30
                ],
                "type": "SimpleItem"
            },
            {
                "AMSLAltAboveTerrain": null,
                "Altitude": 30,
                "AltitudeMode": 1,
                "autoContinue": true,
                "command": 115,
                "doJumpId": 3,
                "frame": 3,
                "params": [
                    340.0,
                    1.0,
                    0.0,
                    0.0,
                    null,
                    null,
                    null
                ],
                "type": "SimpleItem"
            }
        ],
        "plannedHomePosition": [
            47.40298729617688,
            8.5428502,
            473
        ],
        "vehicleType": 2,
        "version": 2
    },
    "rallyPoints": {
        "points": [
        ],
        "version": 2
    },
    "version": 1
}
```

JonasVautherin, 2024-01-08:
Doesn't it confirm that the autopilot does not support the mission item you are trying to use?

mfattori, 2024-01-09:
It would appear so. I've never encountered a message like that before.

JonasVautherin, 2024-01-09:
Ok, closing for now. Please reopen if you realise that it is actually a problem with MAVSDK :+1:.

closed 2024-01-09 by JonasVautherin