mavlink/MAVSDK-Python #650: Communicate with other drones (including broadcasting)

crazymxm, 2024-02-02:
How to communicate with other drones (including broadcasting)?
I saw an example about send_status_text, but it was an example of sending messages to the ground station. I'm wondering if it's possible to broadcast messages to other drones or send messages point-to-point.

crazymxm, 2024-02-02:
I would like to know if my understanding is correct: Each drone has its own system ID in Mavlink, so they can communicate with each other. However, regarding how to broadcast and receive messages, unfortunately, I haven't come across any examples or APIs specifically  in the MAVDdk Python documentation.

julianoes, 2024-02-02:
Basically, messages that don't contain the fields `target_system` and `target_component` or have these both set to 0, are "broadcast" messages. This means that any mavlink router or node is supposed to forward such messages to any system/component.

Taking a step back, what are you trying to achieve? If I have some context, I might be able to help you better.

crazymxm, 2024-02-02:
Thank you for your response.  In my case, I what to report the location to others.
 Yes, Mavlink can broadcast messages. However, let's focus on the Python SDK. Which class and API can be used to send broadcast messages?    is it send_status_text(type, text)?

JonasVautherin, 2024-02-02:
> Mavlink can broadcast messages. However, let's focus on the Python SDK. Which class and API can be used to send broadcast messages? is it send_status_text(type, text)?

MAVSDK sends MAVLink messages. Some messages are meant to be broadcasted, and in that case MAVSDK broadcasts them. Some are targeted messages, and MAVSDK treats them accordingly.

So the real question is: what is it that you are trying to do, and does MAVSDK-Python provide what you want? But we can't guess what you are trying to do :see_no_evil:.

crazymxm, 2024-02-03:
> So the real question is: what is it that you are trying to do, and does MAVSDK-Python provide what you want? But we can't guess what you are trying to do 🙈.

Hi, I have a group of about 4 drones and want to broadcast their locations to each other.

JonasVautherin, 2024-02-04:
I assume that you can connect a ground station (e.g. QGC) to each one of your drones. Let's say for this example that it is using UDP, and the drone broadcasts MAVLink on 14550. It is trivial to run MAVSDK-Python on the ground and connect to one drone, right?

Now say you want MAVSDK-Python on the ground to connect to all 4 drones. There I see at least two solutions:

1. You have each drone broadcast on a different port (e.g. 14550, 14551, 14552, 14553), and on the ground you run one instance of MAVSDK-Python for each of those ports.
2. You have all the drones broadcast on 14550 by default, but on the ground you have some proxy that discriminates the messages by IP (each drone has its own IP) and forwards the messages on different ports (e.g. 14551, 14552, 14553, 14554). Again, you run one instance of MAVSDK-Python for each of those ports (so the proxy listens to 14550, and the mavsdk instances to 14551-14554).

This way you can connect to all 4 drones with MAVSDK-Python from the ground. But instead of the ground, you could run a similar setup on one of the drone and have it connect to the other 3, right? And you can then extend that to having each drone connect to the other 3.

Would that work for your use-case?