mavlink/MAVSDK-Python #651: 1.4..9 to 2.0.0

Jai-GAY, 2024-02-13:
Hi

$ pip3 install mavsdk==2.0.0 --upgrade
$ sudo pip3 install mavsdk==2.0.0 --upgrade

Is there anyone successfully upgraded it to [2.0.0](https://github.com/mavlink/MAVSDK-Python/releases/tag/2.0.0)?

ERROR: Could not find a version that satisfies the requirement [mavsdk==2.0.0](https://pypi.org/project/mavsdk/)
(from versions: 0.5.2, 0.7.0, 0.8.0, 0.8.1, 0.8.2, 0.9.0, 0.9.1, 0.10.0, 0.11.0, 0.12.0, 0.13.4, 0.14.0, 0.15.0, 0.16.0, 0.16.1, 0.17.0, 0.18.0, 0.19.0, 0.20.0, 0.21.0, 0.22.0, 0.22.1, 0.22.2, 0.22.3, 0.23.0, 1.0.0, 1.2.0, 1.3.0, 1.4.0, 1.4.1, 1.4.2, 1.4.3, 1.4.4, 1.4.5, 1.4.6, 1.4.7, 1.4.8, 1.4.9)
ERROR: No matching distribution found for mavsdk==2.0.0

$ sudo lsb_release -a
No LSB modules are available.
Distributor ID: Ubuntu
Description: Ubuntu 20.04.6 LTS
Release: 20.04
Codename: focal

julianoes, 2024-02-13:
Working on it, thanks for the note.

Jai-GAY, 2024-02-14:
thank you. [2.0.1 installation is okay for both Ubuntu 20.04 and 22.04](https://github.com/mavlink/MAVSDK-Python/releases/tag/2.0.1). but server 2.2.0 behavior is not working. 1.4.9 with server 1.4.7 is okay.

Arducopter v4.2.3

$ /usr/local/lib/python3.10/dist-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyAMA1:921600
[02:56:13|Info ] MAVSDK version: v2.2.0 (mavsdk_impl.cpp:26)

$ /usr/local/lib/python3.10/dist-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyAMA1:921600
[02:59:37|Info ] MAVSDK version: v1.4.17 (mavsdk_impl.cpp:20)
[02:59:37|Info ] Waiting to discover system on serial:///dev/ttyAMA1:921600... (connection_initiator.h:20)
[02:59:37|Debug] New: System ID: 42 Comp ID: 190 (mavsdk_impl.cpp:496)
[02:59:37|Debug] Component Unsupported component (190) added. (system_impl.cpp:377)
[02:59:37|Debug] Discovered 1 component(s) (system_impl.cpp:578)
[02:59:37|Info ] System discovered (connection_initiator.h:63)
[02:59:37|Info ] Server started (grpc_server.cpp:53)
[02:59:37|Info ] Server set to listen on 0.0.0.0:50051 (grpc_server.cpp:54)
[02:59:37|Debug] New: System ID: 1 Comp ID: 1 (mavsdk_impl.cpp:496)
[02:59:37|Debug] Component Autopilot (1) added. (system_impl.cpp:377)
[02:59:37|Debug] Component Camera 1 (100) added. (system_impl.cpp:377)
[02:59:37|Debug] Component Unsupported component (250) added. (system_impl.cpp:377)
[02:59:37|Warn ] Vehicle type changed (new type: 13, old type: 0) (system_impl.cpp:225)
[02:59:37|Debug]

julianoes, 2024-02-14:
Oh, what architecture is that?

Jai-GAY, 2024-02-15:
> Oh, what architecture is that?

Pi 4B 4GB RAM Ubuntu 22.04.
Arducopter with CubeOrange or CubeBlack flight controller Unit.

julianoes, 2024-02-15:
Thanks @Jai-GAY. I will try to reproduce it.

julianoes, 2024-02-15:
> $ /usr/local/lib/python3.10/dist-packages/mavsdk/bin/mavsdk_server serial:///dev/ttyAMA1:921600
> [02:56:13|Info ] MAVSDK version: v2.2.0 (mavsdk_impl.cpp:26)

Does it crash or just stop?

Jai-GAY, 2024-02-16:
> Does it crash or just stop?

did not crash, just exited and stopped.

Jai-GAY, 2024-02-16:
> Ok, so it might likely be the same as #654.

No, I do not see the second line printout. His timing was quite fast from the first to the second printout.
Mine, I waited for a few minutes, but I did not see the second line printout, which had already exited.

[09:38:36|Info ] Waiting to discover system on serial:///dev/serial0:921600... (connection_initiator.h:20)
[09:38:36|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:737)
[09:38:36|Debug] Component Autopilot (1) added. (system_impl.cpp:388)
[09:38:36|Debug] MAVLink: info: GCS connection regained	 (system_impl.cpp:256)

julianoes, 2024-02-16:
Let's see if https://github.com/mavlink/MAVSDK/pull/2223 fixes it. Otherwise, we'll have to do more debugging.

closed 2024-02-19 by julianoes