mavlink/MAVSDK-Python #653: About using multiple 'async for' loops

WeonjiChoi, 2024-02-14:
https://github.com/mavlink/MAVSDK-Python/blob/5245d5c41e8afd772a2095c2f38357ae6f86996e/examples/px4_ev_automation.py#L31-L45

Dear, 

When I run the above example code,
it return **only flight_mode value**, but not health, battery, and else. 

I solve this problem by using 
 - casacade for loops (not recommneded) 
 or 
- `'await'` and `'.__aiter__().__anext__()'`.

Please let me know if I'm wrong or if there is other good way to solve the problem.
Thank you :)

Weonji

julianoes, 2024-02-14:
@JonasVautherin one for you :smile:

JonasVautherin, 2024-02-14:
And `async for` is not running "in the background", it's blocking just like a normal for. If you want to run multiple in parallel, you have to run them in coroutines (just like you would run them in threads in another language). Have a look at the [telemetry example](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/telemetry.py).

WeonjiChoi, 2024-02-14:
Thank you for your explanation. 
But then it would be good if the code has some comments about the point you've been explained. For now, some parts of the code are not reachable. 
Thank you anyway. 😊

closed 2024-02-14 by WeonjiChoi

JonasVautherin, 2024-02-14:
What is not reachable?