mavlink/MAVSDK-Python #654: Illegal instruction (core dumped)

ilysec, 2024-02-15:
Hi there, 

I am trying to run 
```
~/.local/lib/python3.10/site-packages/mavsdk/bin/mavsdk_server serial:///dev/serial0:921600
```

And I get the following result:
```
[09:38:36|Info ] MAVSDK version: v2.2.0 (mavsdk_impl.cpp:26)
[09:38:36|Info ] Waiting to discover system on serial:///dev/serial0:921600... (connection_initiator.h:20)
[09:38:36|Debug] New system ID: 1 Comp ID: 1 (mavsdk_impl.cpp:737)
[09:38:36|Debug] Component Autopilot (1) added. (system_impl.cpp:388)
[09:38:36|Debug] MAVLink: info: GCS connection regained	 (system_impl.cpp:256)
[09:38:36|Debug] MAVLink: critical: Preflight Fail: Crash dumps present on SD, vehicle needs service (system_impl.cpp:256)
[09:38:36|Warn ] Vehicle type changed (new type: 2, old type: 0) (system_impl.cpp:230)
[09:38:36|Debug] Discovered 1 component(s) (system_impl.cpp:563)
[09:38:36|Info ] System discovered (connection_initiator.h:62)
Illegal instruction (core dumped)
```

I tried reinstalling grpcio as mentioneed in https://github.com/mavlink/MAVSDK-Python/issues/345:
```
pip3 uninstall grpcio
sudo apt-get install python3-grpcio
```

However, nothing seems to solve the "Illegal Instruction" problem.

julianoes, 2024-02-15:
Ok. This is on what hardware and what OS?

ilysec, 2024-02-15:
```
PRETTY_NAME="Ubuntu 22.04.3 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.3 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy
```

Holybro Pixhawk RPi CM4 Baseboard

julianoes, 2024-02-15:
Aha, thanks!

julianoes, 2024-02-16:
Fix coming: https://github.com/mavlink/MAVSDK/pull/2223

closed 2024-02-19 by julianoes