mavlink/MAVSDK-Python #656: Manual Actuation with 4 Motors

ilysec, 2024-02-20:
Hello, 

I have a [Holybro Pixhawk RPi CM4 Baseboard](https://docs.px4.io/main/en/companion_computer/holybro_pixhawk_rpi_cm4_baseboard.html), with 4 motors connected to the FMU PWM OUT 1-4 ports and the RPi running Ubuntu 22.04.3 LTS (Jammy Jellyfish). When I use the sliders in the QGC Actuators menu, all motors work and are configured properly.

[image]

However, when I try to run the python script below setting manual values, even though I get the correct output on the terminal, no actuation occurs on the actual drone.

```python
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/serial0:921600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            break

    # set the manual control input after arming
    await drone.manual_control.set_manual_control_input(
        float(0.5), float(0.5), float(0.5), float(0.5)
    )

    # start manual control
    print("-- Starting manual control")
    await drone.manual_control.start_position_control()

    await asyncio.sleep(2)

    print("-- The end!")

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())

```

[image]

Also, when I try to run the following, I get the error below:

```python
#!/usr/bin/env python3

import asyncio
from mavsdk import System

async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/serial0:921600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            break

    await drone.action.set_actuator(1, 0.5)

    await asyncio.sleep(2)

    print("-- The end!")

if __name__ == "__main__":
    # Run the asyncio loop
    asyncio.run(run())

```

```
Exception has occurred: ActionError
TIMEOUT: 'Timeout'; origin: set_actuator(); params: (1, 0.5)
  File "/home/ubuntu/Downloads/MAVSDK-Python/mavsdk/action.py", line 663, in set_actuator
    raise ActionError(result, "set_actuator()", index, value)
  File "/home/ubuntu/Downloads/MAVSDK-Python/examples/wow_test_1.py", line 18, in run
    await drone.action.set_actuator(1, 0.5)
  File "/home/ubuntu/Downloads/MAVSDK-Python/examples/wow_test_1.py", line 36, in <module>
    asyncio.run(run())
mavsdk.action.ActionError: TIMEOUT: 'Timeout'; origin: set_actuator(); params: (1, 0.5)
```

I tried all indices from 1 to 16, but I get the same timeout error.

JonasVautherin, 2024-02-21:
> mavsdk.action.ActionError: TIMEOUT: 'Timeout'

This suggests that the MAVLink messages don't reach the drone correctly (or don't come back). Have you tried simple tests like receiving telemetry from the drone? Does that work?

ilysec, 2024-02-21:
Yes, when I run telemetry.py I get:

```
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
Battery: -100.0
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
In air: False
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
GPS info: GpsInfo: [num_satellites: 0, fix_type: NO_GPS]
```

ilysec, 2024-02-21:
I get the following when trying to run takeoff_and_land.py:

```
ubuntu@ubuntu:~/Downloads/MAVSDK-Python/examples$ python3 takeoff_and_land.py
Waiting for drone to connect...
-- Connected to drone!
Waiting for drone to have a global position estimate...
-- Global position estimate OK
-- Arming
Traceback (most recent call last):
  File "/home/ubuntu/Downloads/MAVSDK-Python/examples/takeoff_and_land.py", line 50, in <module>
    asyncio.run(run())
  File "/usr/lib/python3.10/asyncio/runners.py", line 44, in run
    return loop.run_until_complete(main)
  File "/usr/lib/python3.10/asyncio/base_events.py", line 649, in run_until_complete
    return future.result()
  File "/home/ubuntu/Downloads/MAVSDK-Python/examples/takeoff_and_land.py", line 27, in run
    await drone.action.arm()
  File "/home/ubuntu/Downloads/MAVSDK-Python/mavsdk/action.py", line 328, in arm
    raise ActionError(result, "arm()")
mavsdk.action.ActionError: COMMAND_DENIED: 'Command Denied'; origin: arm(); params: ()
^CException ignored in: <module 'threading' from '/usr/lib/python3.10/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1567, in _shutdown
    lock.acquire()
KeyboardInterrupt: 
Exception ignored in: <function System.__del__ at 0xffff9875d990>
Traceback (most recent call last):
  File "/home/ubuntu/Downloads/MAVSDK-Python/mavsdk/system.py", line 86, in __del__
  File "/home/ubuntu/Downloads/MAVSDK-Python/mavsdk/system.py", line 122, in _stop_mavsdk_server
```

JonasVautherin, 2024-02-21:
> mavsdk.action.ActionError: COMMAND_DENIED

Ok, this means that the autopilot received the MAVLink message, and answered with a "denied" code (presumably because it is not ready to arm). So communications should be fine.

Back to your original question, it seems like `set_actuator()` times out, which suggests that MAVSDK sends a command to the autopilot and never receives an answer. Are you sure that your autopilot supports that command? Is it PX4 or Ardupilot? I am not sure if that was implemented for Ardupilot (and how different it would be from PX4) :thinking:.

ilysec, 2024-02-21:
I am running PX4. I do not have a radio controller, do I have to bypass the radio check? For manual control or offboard control, I don't think so, right?

ilysec, 2024-02-21:
No I actually could run manual_control.py, but it only works sometimes. When it doesn't work, I get timeout on arm().

ilysec, 2024-02-21:
I get this when I run manual_control.py and it works

[image]

ilysec, 2024-02-21:
No I didn't, I simple executed the manual_control.py code which should work without an RC I believe?

JonasVautherin, 2024-02-21:
I would check COM_RC_IN_MODE, I think :thinking:.

ilysec, 2024-02-22:
Will check, thanks!

julianoes, 2024-02-22:
@ilysec in the first example with `manual_control` you did not actually arm.

ilysec, 2024-02-22:
I couldn't arm. I will try to recalibrate the sensors and try again. There seem to be no warnings in manual mode but I still can't arm.

julianoes, 2024-02-22:
Right, but if you can't arm, then motors won't spin, except using the test command.

ilysec, 2024-02-23:
How about this? I can now arm the drone, but cannot stop or disarm. Both give the error "Command Denied". I get that disarm would require the drone to land. But why can't I stop offboard control? Also If I try to write manual control functions or use set_actuator after the arming line, I still get Timeouts.
```python
async def run():

    drone = System()
    await drone.connect(system_address="serial:///dev/serial0:921600")

    print("Waiting for drone to connect...")
    async for state in drone.core.connection_state():
        if state.is_connected:
            print("-- Connected to drone!")
            break
    
    await drone.offboard.set_attitude(Attitude(roll_deg=0.0, pitch_deg=0.0, yaw_deg=0.0, thrust_value=1.0))
    await drone.offboard.start()
    await drone.action.arm()

    await asyncio.sleep(5)

    await drone.action.disarm()
    await drone.offboard.stop()

    print("-- The end!")
```

Also, how do I use ActuatorControl and ActuatorControlGroup correctly? I will run it in Offboard mode if that's the only thing I can do... But I need to control each motor separately. I assume it is not like I did below because nothing happens. I found no examples online to check...
```python
    await drone.offboard.set_attitude(Attitude(roll_deg=0.0, pitch_deg=0.0, yaw_deg=0.0, thrust_value=0.0))
    await drone.offboard.start()
    motor_values = ActuatorControl([ActuatorControlGroup([0.5, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]), 
                                    ActuatorControlGroup([0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0])])
    await drone.offboard.set_actuator_control(motor_values)
    await asyncio.sleep(10)
```

closed 2024-02-26 by ilysec