mavlink/MAVSDK-Python #659: Why can't my python code connect to Mavsdk_server with IP added?

boom6314, 2024-02-27:
When I open two Mavsdk_servers, one with IP and one without IP, they can connect to each other. However, when I use python code to connect to the Mavsdk_server with IP, there is no response.

[image]
[image]

```python=
 import asyncio
 from mavsdk import System 

 async def run():
     drone = System(mavsdk_server_address="localhost", port=50051)
     print("123")
     await drone.connect()
     print("456")
    
   
 if __name__ == "__main__":
     asyncio.run(run())
```

I think I'm stuck at drone.connect because I only see 123

I wonder what kind of problem this is and how to solve it

julianoes, 2024-02-29:
I don't think all the pieces are in place to use server functionality in Python, so it might be tricky.

One problem I see is that both are waiting for a drone, right? So that requires the compid to be 1.

JonasVautherin, 2024-02-29:
MAVSDK-Python will listen on 14540 by default, but you apparently send on 14550. Can you try `mavsdk_server_address="udp://:14550"`?

boom6314, 2024-03-03:
> One problem I see is that both are waiting for a drone, right? So that requires the compid to be 1.

Thank you for your response. Yes, our goal is to use python code to simulate a drone and throw out information through telemetry_server in MAVSDK-Python. We are currently using the simplest method to test the connection, but our Python code cannot connect to Mavsdk_server

boom6314, 2024-03-03:
> MAVSDK-Python will listen on 14540 by default, but you apparently send on 14550. Can you try mavsdk_server_address="udp://:14550"?

Thank you for your response.I tried it but the code still only shows 123

```python =
import asyncio
from mavsdk import System 

async def run():
    drone = System(mavsdk_server_address="udp://:14550", port=50051)
    print("123")
    await drone.connect()
    print("456")
   ```

JonasVautherin, 2024-03-03:
And what is on the other side?

JonasVautherin, 2024-03-04:
Are you sure that your python script does not interfere with your server? Seems like both will try to bind to 50051. On the screenshots below, the "listener" is on `-p 50055`, but your python script uses `50051`. Is that expected?

boom6314, 2024-03-05:
> Are you sure that your python script does not interfere with your server? Seems like both will try to bind to 50051.

According to the instructions of Mavsdk, the Mavsdk_server with IP added will throw out the information. So my idea is to set the port of the python code to be the same as the Mavsdk_server with added IP, and let the python code control what information my Mavsdk_server will throw out.
So first I want to try to use python code to connect to Mavsdk_server with IP added.

JonasVautherin, 2024-03-05:
What I mean is that your screenshots have two setups:

* -p 50055 udp://:14550
* -p 50051 udp://127.0.0.1:14550

Your Python code does not do one or the other, but a mix of them: -p 50051 udp://:14550.

boom6314, 2024-03-06:
> What I mean is that your screenshots have two setups:
Sorry for taking a while to reply

When I use ```mavsdk_server_address='udp://127.0.0.1:14550',port=50051 ``` that code only show 123
```python 
import asyncio
from mavsdk import System 

async def run():
    drone = System(mavsdk_server_address='udp://127.0.0.1:14550',port=50051)
    print("123")
    await drone.connect()
    print("456")
   
if __name__ == "__main__":
    asyncio.run(run())
```
When I use ```port=50055 ``` and that code can show 123 and 456
```python
import asyncio
from mavsdk import System 

async def run():
    drone = System(port=50055)
    print("123")
    await drone.connect()
    print("456")
   
if __name__ == "__main__":
    asyncio.run(run())
```
Does this mean that my python code can connect to Mavsdk_server without IP but cannot connect to Mavsdk_server with IP?

boom6314, 2024-03-15:
> I don't think all the pieces are in place to use server functionality in Python, so it might be tricky.

I finally used a tricky method. I first set the IP of Mavsdk_server to ```127.0.0.1```, connected it to my python code on the local side, and then used ```Cmavnode``` to forward the information to the target IP.