mavlink/MAVSDK-Python #660: Update to RTK example

Pjort, 2024-02-27:
In the comments for the RTK example code, it states "With MAVSDK v2, the API will change to a vector of bytes instead of this clunky string".

I was wondering instead of making into a string, should the rtcm_correction_data just be passed directly to the send_rtcm_data? Or should it still be inside a rtk.RtcmData?

Reference:
[https://github.com/mavlink/MAVSDK-Python/blob/main/examples/rtk.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/rtk.py)

julianoes, 2024-02-29:
Oh, good point. We need to update this now that v2 is out. You should try without `str` and report back if you get it to work.

Pjort, 2024-05-03:
When I remove str I get the following error:

ValueError: b'\xd3\x00iF@\x00m\xa7J\x02\x00\x00\x00\xac\x00\xe0\x00\x00\x00\x00 \x02\x00\x00\x7f\xd5\x05\x0e\x9d\x06\x8e\x9c\xa1\xbb\xaf\x1a\xe6\xaf\x08Y^N\x9b\xec\xe4\xff\x1f}Kk\x1d\xd1S\x84\xbb\x01\x08\xfe\xa0\xffa\xfa}:\xb0\xf3\xab3\xd4\x16\x8f\xa2\xcb\xbdn\xa6\xf4F\xfb\xc2\xdf\xae\xc4\x91\x01\xf9\x89\x03\xe8\xc0\x0e\xde_\xf9\xff\xff\xff\xff\x00\x12\xd2\xe45S4\xb8[HO>\x10' has type str, but isn't valid UTF-8 encoding. Non-UTF-8 strings must be converted to unicode objects before being added.

Pjort, 2024-05-03:
When I swap back to MAVSDK version: v1.4.18 everything once again works fine.

Pjort, 2024-05-03:
It seems that the problem is related to this hacky fix that was back in 1.4.16:
[https://github.com/mavlink/MAVSDK/pull/2073/commits/9d02d742ac00633194804916b30563985e5926a3](https://github.com/mavlink/MAVSDK/pull/2073/commits/9d02d742ac00633194804916b30563985e5926a3)

But the new version MAVSDK-Python doesn't reflec what was mentioned with using a vector<uint8_t> instead.

julianoes, 2024-05-04:
This is broken. I need to update it to base64. I have just recently added support for base64 in MAVSDK, so that can now happen.

julianoes, 2024-05-04:
If this is urgent and blocking you, consider [putting something in the hat](https://github.com/sponsors/julianoes).

Jsmitx, 2024-05-06:
How big an issue is it to fix it? It could be a cool feature.

rgreenaus, 2024-07-22:
This is a big blocker for me. Could we please get this one prioritised?

julianoes, 2024-07-23:
Yes, @rgreenaus. It will be out as soon as v3 is done, see https://github.com/mavlink/MAVSDK/issues/2316.

Alternatively, you could try to build the server yourself already, and use that manually but I'm not 100% sure it will all work out.

rgreenaus, 2024-07-24:
Awesome, thank you. In the mean time I have gotten it working by compiling the mavsdk_server from master and just did a drop-in replacement with the binary from mavsdk-python 2.8

The only changes needed in the python code (this will work for the examples too) were:

+ import base64

-await drone.rtk.send_rtcm_data(rtk.RtcmData(str(rtcm_correction_data)))
+await drone.rtk.send_rtcm_data(rtk.RtcmData(base64.b64encode(rtcm_correction_data)))

Cheers

julianoes, 2024-07-24:
Yes that seems right! I'm glad you are (somewhat) unblocked @rgreenaus.