mavlink/MAVSDK-Python #662: get_entries() gets stuck

M4ster00gway, 2024-03-05:
When attempting to download logs after conducting multiple, relatively long flights, I am encountering a problem where the get_entries() function becomes unresponsive. It neither returns any log entries nor generates any errors. I have tried using MAVSDK versions 2.1.0 and 1.4.9 and i use the PX4 autopilot system.

```python

async def download_log_file(drone, entry):
    date_without_colon = entry.date.replace(":", "-")
    filename = f"./log-{date_without_colon}.ulog"
    print(f"Downloading: log {entry.id} from {entry.date} to {filename}")
    previous_progress = -1
    async for progress in drone.log_files.download_log_file(entry, filename):
        new_progress = round(progress.progress * 100)
        if new_progress != previous_progress:
            sys.stdout.write(f"\r{new_progress} %")
            sys.stdout.flush()
            previous_progress = new_progress
    print("\nDownload completed.")

async def download_all_logs(drone):
    entries = await drone.log_files.get_entries()
    if not entries:
        print("No log entries found.")
        return

    download_tasks = [download_log_file(drone, entry) for entry in entries]
    await asyncio.gather(*download_tasks)
    ```
    print(f"Downloading: log {entry.id} from {entry.date} to {filename}")
    previous_progress = -1
    async for progress in drone.log_files.download_log_file(entry, filename):
        new_progress = round(progress.progress * 100)
        if new_progress != previous_progress:
            sys.stdout.write(f"\r{new_progress} %")
            sys.stdout.flush()
            previous_progress = new_progress
    print("\nDownload completed.")

async def download_all_logs(drone):
    entries = await drone.log_files.get_entries()
    if not entries:
        print("No log entries found.")
        return

    download_tasks = [download_log_file(drone, entry) for entry in entries]
    await asyncio.gather(*download_tasks)
    ```

This is where it gets stuck:

```python
entries = await drone.log_files.get_entries()
```

Any ideas on how to solve this?

julianoes, 2024-03-06:
Thanks for the issue. Does it work in QGC?

julianoes, 2024-03-07:
@M4ster00gway could yes. Worth trying again. We just need to push a mavsdk-python update.

julianoes, 2024-03-07:
https://github.com/mavlink/MAVSDK-Python/pull/664

julianoes, 2024-03-07:
Try https://github.com/mavlink/MAVSDK-Python/releases/tag/2.1.0

M4ster00gway, 2024-03-14:
Thanks and sorry for the late reply! @julianoes 

Getting the entries works better now, however after retrieving the entries, mavsdk throws this error:

```bash
terminate called after throwing an instance of 'std::future_error'
  what():  std::future_error: Promise already satisfied
```

Which causes the mavsdk server to crash so i cant download the logs.

julianoes, 2024-03-19:
Any chance you can get a backtrace using gdb?

julianoes, 2024-05-20:
I'm assuming this has been fixed. If it still exists, let me know.

closed 2024-05-20 by julianoes