mavlink/MAVSDK-Python #666: goto_location() unsupported error

l0ad-pri0r, 2024-03-13:
Hello everyone!

I developed a script that was tested in a SITL environment. Everything was working as expected but when I tried to test it in a real drone I receive the UNSUPPORTED error on the goto_location. 
I'am using it just to change the drone altitude. So firstly I get the current drone position and use it on the first two parameters. For the altitude make some simple math, and for the yaw I use 'float(nan)':

`await drone.action.goto_location(current_pos['latitude'], current_pos['longitude'], (current_pos['abs_altitude'] - (current_pos['altitude'] - altitude)), float('nan'))`

This instruction returns the error before mentioned. So I tried to pass direct values casted to the especified type, and still the same error... No errors are shown on QGC console. And again, in a simulation environment it works as excpected.

What can cause his behaviour? Thanks in advance! 🙂

julianoes, 2024-03-19:
Ok, is this PX4 or ArduPilot?

l0ad-pri0r, 2024-03-19:
PX4, using the latest stable firmware version

julianoes, 2024-03-19:
When you try SITL, I assume that's not latest stable/v1.14?

My guess is you land right here:
https://github.com/PX4/PX4-Autopilot/blob/release/1.14/src/modules/commander/Commander.cpp#L703-L704

It wants two things:
1. mode switch set, which we have, see https://github.com/mavlink/MAVSDK/blob/main/src/mavsdk/plugins/action/action_impl.cpp#L459.
2. no unsupported bits set, which we don't do.

That leaves me confused.