mavlink/MAVSDK-Python #667: How to use Param_server and Action_server

boom6314, 2024-03-15:
I created a python code to simulate the UAV transmitting battery data to our own GCS. I used param_server to provide ```'CAL_GYRO0_ID'```,``` 'CAL_ACC0_ID'```,``` 'CAL_MAG0_ID' ```and ```'SYS_HITL'```, but my Terminal still popped up Missing Param message.Is there anything I missed in the code?
``` python
import asyncio
from mavsdk import System
from mavsdk.telemetry_server import (Battery)
from mavsdk.param_server import  (ParamServer)
async def publish_battery():

 drone = System(port=50053) 
 await drone.connect("udp://:14550") 
 
 await drone.param_server.provide_param_int('CAL_GYRO0_ID', 1)
 await drone.param_server.provide_param_int('CAL_ACC0_ID', 1)
 await drone.param_server.provide_param_int('CAL_MAG0_ID', 1)
 await drone.param_server.provide_param_int('SYS_HITL', 0)  

 while True:
    remaining_percent = 0.8 # float 0.0~1.0
    voltage_v = 12.0        # float
    await drone.telemetry_server.publish_battery(Battery(voltage_v,remaining_percent))
    await asyncio.sleep(1)
if __name__ == "__main__":
 loop = asyncio.get_event_loop()
 loop.run_until_complete(publish_battery())  
 ```
 
[image]

In addition, I also want to know how to use python code to call the Action_server plugin so that I can control my homemade flight controller.

julianoes, 2024-03-18:
Creating a server hasn't really been tested using Python, I think. 

@JonasVautherin you haven't tried this either, right?

JonasVautherin, 2024-03-18:
No I haven't. Only in C++. Maybe something is missing for the language bindings? :thinking:

boom6314, 2024-03-19:
> Creating a server hasn't really been tested using Python, I think.

But when I use ```telemetry_server``` I can send my battery information to my GCS correctly. So this means that some servers are not complete, right?
I use MAVSDK version: ```v1.4.17```, can this problem be solved if I use ```v2.1```?

julianoes, 2024-03-19:
Yes, please use v2. There have been a lot of changes and fixes for better server support.

boom6314, 2024-03-21:
Ok, I will try it with v2, thank you for your replies