mavlink/MAVSDK-Python #668: health.is_global_position_ok and health.is_home_position_ok with Ardupilot

Jai-GAY, 2024-03-15:
this line of code does not work with Ardupilot. It never comes out from the for loop. MissionPlanner reports GPS is healthy, and Ardupilot acquired a home position. 

[if health.is_global_position_ok and health.is_home_position_ok:](https://github.com/mavlink/MAVSDK-Python/blob/0d137ca68455a7bb4cd82996899e9290fcb6d2ef/examples/telemetry_takeoff_and_land.py#L46)

[telemetry.Health](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/telemetry.html#mavsdk.telemetry.Health)

Pi4B Ubuntu 22.04 64-bit
Arducopter v4.2.3
Name: mavsdk
Version: 2.1.0
Summary: Python wrapper for MAVSDK
Home-page: https://github.com/mavlink/MAVSDK-Python
Author:
Author-email:
License: UNKNOWN
Location: /home/pi/.local/lib/python3.10/site-packages
Requires: aiogrpc, grpcio, protobuf
Required-by:
$

arpitpara, 2024-03-16:
Try printing what is the response of health from telemetry:
    async for health in drone.telemetry.health():
        print(health,'Drone Health')
        if health.is_global_position_ok and health.is_home_position_ok:
            print("-- Global position state is good enough for flying.")

It checks for your Home position is set or not with GPS health.

Jai-GAY, 2024-03-16:
> Try printing what is the response of health from telemetry:

If I don't break from the for loop, I don't see anything printing. I can only do once or I try to do a counter.

arpitpara, 2024-03-16:
Actually you should be getting some response printing cuz everytime you ask for response telemetry it gives you something.. if condition only is for checking if it's ok or not. 
you can put a break statement after and can only run it for once. Just for testing that only this statement is a problem or not.

JonasVautherin, 2024-03-16:
If it hangs there, it may mean that it never receives the message it expects. Is there a different MAVLink message between PX4 and Ardupilot here?

Jai-GAY, 2024-03-18:
From this [doc](https://mavsdk.mavlink.io/main/en/cpp/api_reference/structmavsdk_1_1_telemetry_1_1_health.html), it expects a boolean, my guess is all are false always.

> Is there a different MAVLink message between PX4 and Ardupilot here?

Not sure, [Copter Mission Command List](https://ardupilot.org/copter/docs/mission-command-list.html#mission-command-list).

Jai-GAY, 2024-03-18:
> Try printing what is the response of health from telemetry:

is the is_global_position_ok that failed it. But is_home_position_ok is okay, interesting.

Main in
System server b4
System server done
Pi4B Waiting for drone to connect...
-- Connected to drone!
Version: [flight_sw_major: 4, flight_sw_minor: 2, flight_sw_patch: 3, flight_sw_vendor_major: 56, flight_sw_vendor_minor: 52, flight_sw_vendor_patch: 97, os_sw_major: 0, os_sw_minor: 0, os_sw_patch: 0, flight_sw_git_hash: 376130633 , os_sw_git_hash: 663466323230383 , flight_sw_version_type: RELEASE]
Waiting for drone to have a global position estimate...
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, **is_global_position_ok: False**, **is_home_position_ok: True**, is_armable: False]  i =  0
-- Global position state NOT good for flying. i =  0
Battery: 9800.0
Battery_v: 15.52500057220459
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  1
-- Global position state NOT good for flying. i =  1
Battery: 9800.0
Battery_v: 15.528000831604004
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, **is_global_position_ok: False**, **is_home_position_ok: True**, is_armable: False]  i =  2
-- Global position state NOT good for flying. i =  2
Battery: 9800.0
Battery_v: 15.481000900268555
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  3
-- Global position state NOT good for flying. i =  3
Battery: 9800.0
Battery_v: 15.533000946044922
Battery: 9800.0
Battery_v: 15.533000946044922
Battery: 9800.0
Battery_v: 15.533000946044922
print_status_text CancelledError
Battery: 9800.0
Battery_v: 15.527000427246094
Battery: 9800.0
Battery_v: 15.533000946044922
Battery: 9800.0
Battery_v: 15.53700065612793
print_battery CancelledError
-- run ended
Main out

arpitpara, 2024-03-18:
Yeah !!
Your GPS needs to be fixed. check if you can arm it through MP or try doing GPS calibration. 
' is_global_position_ok: False,'
This is the indicator that it doesn't get any GPS location. You can try getting your GPS info printed through telemetry through another example, 'examples/telemetry.py'.

Jai-GAY, 2024-03-18:
> Your GPS needs to be fixed. check if you can arm it through MP or try doing GPS calibration.

GPS already re-calibrated.
MP already reported 3D Fix.
I can arm with loiter flight mode and fly.

> 'examples/telemetry.py'

Okay, will take a look at the code.

Jai-GAY, 2024-03-18:
> You can try getting your GPS info printed through telemetry through another example, '[examples/telemetry.py](https://github.com/mavlink/MAVSDK-Python/blob/main/examples/telemetry.py)'.

likely found the problem, not working.

I think something similar to this is missing.
await drone.telemetry.set_rate_battery(1)

print_position CancelledError
print_gps_info CancelledError

**** test code function ***
async def print_gps_info(drone):
    try:
        async for gps_info in drone.telemetry.gps_info():
            print(f"GPS info: {gps_info}")
    except asyncio.CancelledError:
        print("print_gps_info CancelledError")
        return  # pass

async def print_position(drone):
    try:
        async for position in drone.telemetry.position():
            print(position)
    except asyncio.CancelledError:
        print("print_position CancelledError")
        return  # pass
**** test code function ***

Main in
System server b4
System server done
Pi4B Waiting for drone to connect...
-- Connected to drone!
Version: [flight_sw_major: 4, flight_sw_minor: 2, flight_sw_patch: 3, flight_sw_vendor_major: 56, flight_sw_vendor_minor: 52, flight_sw_vendor_patch: 97, os_sw_major: 0, os_sw_minor: 0, os_sw_patch: 0, flight_sw_git_hash: 376130633 , os_sw_git_hash: 663466323230383 , flight_sw_version_type: RELEASE]
Waiting for drone to have a global position estimate...
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  0
-- Global position state NOT good for flying. i =  0
Battery: 9700.0
Battery_v: 15.399001121520996
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  1
-- Global position state NOT good for flying. i =  1
Battery: 9700.0
Battery_v: 15.399001121520996
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  2
-- Global position state NOT good for flying. i =  2
Battery: 9700.0
Battery_v: 15.396000862121582
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  3
-- Global position state NOT good for flying. i =  3
Battery: 9700.0
Battery_v: 15.396000862121582
Battery: 9700.0
Battery_v: 15.395000457763672
Battery: 9700.0
Battery_v: 15.390000343322754
**print_position CancelledError
print_gps_info CancelledError**
print_status_text CancelledError
Battery: 9700.0
Battery_v: 15.402000427246094
Battery: 9700.0
Battery_v: 15.374000549316406
Battery: 9700.0
Battery_v: 15.405000686645508
print_battery CancelledError
-- run ended
Main out

Jai-GAY, 2024-03-18:
subscribe to these, now we have the info. the problem is clear.

    telem_gps = asyncio.ensure_future(print_gps_info(drone))
    await drone.telemetry.set_rate_gps_info(1)
    telem_pos = asyncio.ensure_future(print_position(drone))
    await drone.telemetry.set_rate_position(1)

Main in
System server b4
System server done
Pi4B Waiting for drone to connect...
-- Connected to drone!
Version: [flight_sw_major: 4, flight_sw_minor: 2, flight_sw_patch: 3, flight_sw_vendor_major: 56, flight_sw_vendor_minor: 52, flight_sw_vendor_patch: 97, os_sw_major: 0, os_sw_minor: 0, os_sw_patch: 0, flight_sw_git_hash: 376130633 , os_sw_git_hash: 663466323230383 , flight_sw_version_type: RELEASE]
Battery: 9700.0
Battery_v: 15.351000785827637
Waiting for drone to have a global position estimate...
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  0
-- Global position state NOT good for flying. i =  0
Battery: 9700.0
Battery_v: 15.357000350952148
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  1
-- Global position state NOT good for flying. i =  1
**Position info: Position: [latitude_deg: 1.XXXXXX, longitude_deg: 103.YYYYYY, absolute_altitude_m: 21.830001831054688, relative_altitude_m: -0.010000000707805157]**
**GPS info: GpsInfo: [num_satellites: 15, fix_type: FIX_3D]**
Battery: 9700.0
Battery_v: 15.358000755310059
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  2
-- Global position state NOT good for flying. i =  2
Position info: Position: [latitude_deg: 1.XXXX, longitude_deg: 103.YYYYY, absolute_altitude_m: 21.830001831054688, relative_altitude_m: -0.013000000268220901]
GPS info: GpsInfo: [num_satellites: 15, fix_type: FIX_3D]
Battery: 9700.0
Battery_v: 15.35400104522705
-- Health :  Health: [is_gyrometer_calibration_ok: True, is_accelerometer_calibration_ok: True, is_magnetometer_calibration_ok: True, is_local_position_ok: False, is_global_position_ok: False, is_home_position_ok: True, is_armable: False]  i =  3
-- Global position state NOT good for flying. i =  3
Position info: Position: [latitude_deg: 1.XXXXX, longitude_deg: 103.YYYYYY, absolute_altitude_m: 21.84000015258789, relative_altitude_m: -0.00800000037997961]
GPS info: GpsInfo: [num_satellites: 15, fix_type: FIX_3D]
Battery: 9700.0
Battery_v: 15.36500072479248
Position info: Position: [latitude_deg: 1.XXXXX, longitude_deg: 103.YYYYY, absolute_altitude_m: 21.850000381469727, relative_altitude_m: 0.004000000189989805]
GPS info: GpsInfo: [num_satellites: 15, fix_type: FIX_3D]
Battery: 9700.0
Battery_v: 15.368000984191895
Position info: Position: [latitude_deg: 1.XXXXX, longitude_deg: 103.YYYYYY, absolute_altitude_m: 21.84000015258789, relative_altitude_m: -0.0020000000949949026]
GPS info: GpsInfo: [num_satellites: 15, fix_type: FIX_3D]
Battery: 9700.0
Battery_v: 15.367000579833984
print_position CancelledError
print_gps_info CancelledError
print_status_text CancelledError
Battery: 9700.0
Battery_v: 15.358000755310059
Battery: 9700.0
Battery_v: 15.374000549316406
Battery: 9700.0
Battery_v: 15.342000961303711
print_battery CancelledError
-- run ended
Main out

arpitpara, 2024-03-21:
You might wanna try downgrading your Ardupilot Firmware Stable Version. 4.1 or less.

Jai-GAY, 2024-03-21:
> downgrading your Ardupilot Firmware Stable Version. 4.1 or less.

the latest stable version is 4.4.4, and [4.5.0 is in the last stage of beta testing](https://github.com/ArduPilot/ardupilot/blob/master/ArduCopter/ReleaseNotes.txt) before release. There are many bug fixes and performance improvements.

arpitpara, 2024-03-21:
No it doesn't mean that you're permanently shifting to 4.1. It helps us find out the bug and we might be able to fix it in further releases. Some version upgrades changes the parameters inside the firmware, If the code is running on older version and not new, we can update the code with newer parameters according to Firmware Version.

ilamanov, 2024-04-10:
I am seeing the same issue on the following:
MAVSDK-python version: 2.3.0
Arducopter: 4.5.1

I always get
```
Health: [
is_gyrometer_calibration_ok: True,
is_accelerometer_calibration_ok: True,
is_magnetometer_calibration_ok: True,
**is_local_position_ok: False,**
**is_global_position_ok: False,**
is_home_position_ok: True,
is_armable: False
]
```

julianoes, 2024-04-11:
@ilamanov what's the setup where you test this? SITL, or real copter (outside) with GPS signal?

ilamanov, 2024-04-16:
@julianoes It's a real copter outside with ~17 sats

ShafSpecs, 2024-05-14:
Not sure if this is related but I do get a few telemetry methods hanging with ardupilot:
- is_in_air
- landed_state
- status_text
- mission_progress

Anytime I attempt to print them, it just hangs. Till I manually break out of the loop or shut the program down. Health works well for me though

Jai-GAY, 2024-05-15:
> Anytime I attempt to print them, it just hangs.

Try again with ```print("XXXX", flush=True)```