mavlink/MAVSDK-Python #670: handling ActionError

Jai-GAY, 2024-03-20:
like to know what is the proper way to handle/receive the Raises exception for example
[async arm()](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html#mavsdk.action.Action.arm)
[async takeoff()](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html#mavsdk.action.Action.takeoff)

Is there [an example code](https://github.com/mavlink/MAVSDK-Python/tree/main/examples)?

   try:
        actionerr = await drone.action.takeoff()
        .........
    except [ActionError](http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/action.html#mavsdk.action.ActionError)(result=res, origin=org):
        .......

julianoes, 2024-03-20:
Something like this:

https://github.com/mavlink/MAVSDK-Python/blob/69576ac9ab223317f853f6612a364572f5a8eae8/examples/camera.py#L23-L27

Jai-GAY, 2024-03-21:
Appreciated and got it.
Thank you for the development work and for making the lives of the users easier.

closed 2024-03-21 by Jai-GAY