mavlink/MAVSDK-Python #671: getting mavsdk version

Jai-GAY, 2024-03-21:
Is it the correct way to retrieve it?

print("System server version : ", mavsdk.major, " ", mavsdk.minor)
output
System server version :  3   10

https://github.com/mavlink/MAVSDK/releases
https://github.com/mavlink/MAVSDK-Python/releases

arpitpara, 2024-03-21:
No, The Latest Mavsdk Version available is - 2.1
This might be your python version, Really Not sure about it.
You can check the mavsdk version in your MAVSdk-Python Repo, Through
`cat MAVSDK_SERVER_VERSION`, this will be your downloaded github repo. It sometimes differs to your installed mavsdk server version.
Check through
`sudo apt search libmavsdk-dev`

Jai-GAY, 2024-03-21:
> This might be your python version, Really Not sure about it.

Yes, that is my Python 3 version as well. maybe. Yes, I am very sure my [Mavsdk-python is 2.1.0](https://pypi.org/project/mavsdk/#history).

arpitpara, 2024-03-21:
Ok.
Then You can close the issue if you're able to find the version.

Jai-GAY, 2024-03-21:
> Ok. Then You can close the issue if you're able to find the version.

No, I cannot through Python 3 code.

julianoes, 2024-03-21:
Ok, so we should expose this function via Python:
https://github.com/mavlink/MAVSDK/blob/10876a89b5d4097b68009fa98b30b1ff08c29488/src/mavsdk/core/include/mavsdk/mavsdk.h#L54-L61

I think we need to add it to proto here:
https://github.com/mavlink/MAVSDK-Proto/blob/main/protos/core/core.proto#L9

And then implement it here:
https://github.com/mavlink/MAVSDK/blob/10876a89b5d4097b68009fa98b30b1ff08c29488/src/mavsdk_server/src/core/core_service_impl.h

Pull requests welcome :smile: