mavlink/MAVSDK-Python #675: drone.connect with time out

Jai-GAY, 2024-03-25:
what is the best method to timeout this [connect calling](https://github.com/mavlink/MAVSDK-Python/blob/8ebfdecbd894dac5763349fc49c26ba5323ea128/examples/takeoff_and_land.py#L10C11-L10C24)?

I have tried this but I believe the asyncio is too old and [timeout](https://docs.python.org/3/library/asyncio-task.html#asyncio.timeout) is not implemented yet. I cannot find [what asyncio version is installed](https://blog.finxter.com/how-to-check-asyncio-package-version-in-python/). I can only use [asyncio.wait_for](https://docs.python.org/3/library/asyncio-task.html#asyncio.wait_for), but connection without await is not reliable anymore.

Python 3.10.
Name: mavsdk
Version: 2.1.0
Summary: Python wrapper for MAVSDK
Home-page: https://github.com/mavlink/MAVSDK-Python
Author: 
Author-email: 
License: UNKNOWN
Location: /home/pi/.local/lib/python3.10/site-packages
Requires: aiogrpc, grpcio, protobuf
Required-by:

try:
    # set a timeout
    async with asyncio.timeout(5):
        await drone.connect(system_address="serial:///dev/ttyAMA1:921600")
except asyncio.TimeoutError:
    print(f"Timeout waiting drone.connect")
    return False

AttributeError: module 'asyncio' has no attribute 'timeout

res = await asyncio.wait_for(drone.connect(system_address="serial:///dev/ttyAMA1:921600"), 10)

julianoes, 2024-03-26:
@JonasVautherin ideas?

JonasVautherin, 2024-03-26:
The link to the Python docs says: "New in version 3.11."

@Jai-GAY did you try updating your Python to >= 3.11? :thinking:

closed 2024-03-26 by julianoes

Jai-GAY, 2024-03-27:
Noted.

The Official [jammy 22.04 LTS support](https://packages.ubuntu.com/jammy/python3) is 3.10.x.