mavlink/MAVSDK-Python #678: Running mavsdk server on Windows

ShafSpecs, 2024-04-16:
Hello, regarding using Windows with Mavsdk-python: I am using a separate mavsdk server binary (that I boot up before connecting) due to the absence of the binaries on mavsdk-python-windows-edition, I wanted to ask about startup and shutdown - what's the correct way to actually handle the server? Do I start the binary just before calling `drone.connect()` and shout it down after disconnecting?

And what of a case where I have multiple connections? Do I create up a new server process for each of them? Or just using one executable is fine. Thank you

JonasVautherin, 2024-04-18:
You would have to start `mavsdk_server` manually, once for each drone. Say one drone sends MAVLink to 14550 and the other one to 14551, you would run something like:

```
mavsdk_server -p 50051 udp://:14550
mavsdk_server -p 50052 udp://:14551
```

Then from python you would do something like:

```python
drone1 = System(mavsdk_server_address="127.0.0.1", port=14551)
await drone1.connect()

drone2 = System(mavsdk_server_address="127.0.0.1", port=14552)
await drone2.connect()
```

So that drone1 connects to mavsdk_server on 14551 which listens for MAVLink on 14550, and drone2 connects to the other mavsdk_server on 14552 which listens for MAVLink on 14551.

Does that make sense?

ShafSpecs, 2024-04-18:
> Does that make sense?

Yes, perfectly. Thanks. That solves this issue for me.

> You would have to start `mavsdk_server` manually, once for each drone. Say one drone sends MAVLink to 14550 and the other one to 14551, you would run something like:
> 
> ```
> mavsdk_server -p 50051 udp://:14550
> mavsdk_server -p 50052 udp://:14551
> ```

On another note, if I am using serial, that just means different ports. Except a serial is configured to connect to more than one simultaneously, in that case you pass the same serial port but run mavsdk on different ports, right?

Not part of the query, but do you please have any resource for fleet management (more than one drone)? Being able to communicate amongst themselves as well as with the GCS.

closed 2024-04-18 by ShafSpecs