mavlink/MAVSDK-Python #681: How to manual control

13eav1s, 2024-04-17:
Hello, I can't to manual control the drone I am 100% sure that drone in Manual flight mode

```python
import random

from mavsdk import System
import asyncio

manual_inputs = [
    [0, 0, 0.5, 0],  # no movement
    [-1, 0, 0.5, 0],  # minimum roll
    [1, 0, 0.5, 0],  # maximum roll
    [0, -1, 0.5, 0],  # minimum pitch
    [0, 1, 0.5, 0],  # maximum pitch
    [0, 0, 0.5, -1],  # minimum yaw
    [0, 0, 0.5, 1],  # maximum yaw
    [0, 0, 1, 0],  # max throttle
    [0, 0, 0, 0],  # minimum throttle
]

async def run():
    drone = System(mavsdk_server_address='localhost', port=50051)
    # await drone.connect(system_address='serial:///dev/ttyACM0')
    print('Connecting to drone')

    await drone.connect(system_address='serial:///dev/tty.usbmodem1103')

    status_text_task = asyncio.ensure_future(print_status_text(drone))

    print('Waiting for drone to connect...')
    async for state in drone.core.connection_state():
        if state.is_connected:
            print(f'-- Connected to drone')
            break

    print("-- set manual control values")
    await drone.manual_control.set_manual_control_input(
        float(0), float(0), float(0.5), float(0)
    )

    # start manual control
    print("-- Starting manual control")
    await drone.manual_control.start_position_control()

    while True:
        # grabs a random input from the test list
        # WARNING - your simulation vehicle may crash if its unlucky enough
        input_index = random.randint(0, len(manual_inputs) - 1)
        input_list = manual_inputs[input_index]

        # get current state of roll axis (between -1 and 1)
        roll = float(input_list[0])
        # get current state of pitch axis (between -1 and 1)
        pitch = float(input_list[1])
        # get current state of throttle axis
        # (between -1 and 1, but between 0 and 1 is expected)
        throttle = float(input_list[2])
        # get current state of yaw axis (between -1 and 1)
        yaw = float(input_list[3])

        await drone.manual_control.set_manual_control_input(
            pitch, roll, throttle, yaw)

        await asyncio.sleep(0.1)

    status_text_task.cancel()

async def print_status_text(drone):
    try:
        async for status_text in drone.telemetry.status_text():
            print(f'Status: {status_text.type}: {status_text.text}')
    except asyncio.CancelledError:
        return

if __name__ == '__main__':
    asyncio.run(run())
```

but when I try to but when I try to start_position_control I get the following:
[image]

and in mavsdk server I get this
[image]

husanboyqodirov, 2025-02-21:
While choosing SITL firmware, make sure to choose Multirotor.

JonasVautherin, 2025-02-21:
Feels like a discrepancy between the Python bindings and the mavsdk_server you're using. Which version of MAVSDK-Python are you using, and which version of mavsdk_server?