mavlink/MAVSDK-Python #683: Unable to start a mission a second time

sienks, 2024-04-18:
PX4 version: Latest
MAVSDK-Python version: 2.5.1

**Expected behavior:**
1. Upload mission: drone.mission.upload_mission -> mission is uploaded 
2. Start Mission: drone.mission.start_mission -> mission is performed
3. Start Mission: drone.mission.start_mission-> mission is performed

Step 3 results in the aircraft loitering despite a mission being uploaded. Here is the PX4 log output below.
```
pxh> INFO  [mavlink] partner IP: 127.0.0.1
INFO  [commander] Armed by external command
pxh> INFO  [tone_alarm] arming warning
pxh> INFO  [navigator] Using default takeoff altitude: 2.5 m
pxh> INFO  [commander] Takeoff detected
pxh> INFO  [tone_alarm] notify positive
INFO  [navigator] Executing Mission
pxh> INFO  [navigator] Mission finished, loitering
WARN  [navigator] No valid mission available, loitering
```

Is this a regression? 

I recall an aircraft being able to perform the same mission a second time provided it hadn't been cleared from the flight controller with the clear mission command.

julianoes, 2024-04-19:
Have you tried setting the current mission item to 0 before starting again?

For Mission:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mission.html#mavsdk.mission.Mission.set_current_mission_item

Or MissionRaw:
http://mavsdk-python-docs.s3-website.eu-central-1.amazonaws.com/plugins/mission_raw.html#mavsdk.mission_raw.MissionRaw.set_current_mission_item

sienks, 2024-04-20:
Thanks, setting the current mission item to 0 was the solution.

closed 2024-04-20 by sienks