mavlink/MAVSDK-Python #685: One `mavsdk_server`, multiple drones

ShafSpecs, 2024-04-21:
Hello, I wanted to ask if it's possible to have one mavsdk server instance connected to multiple drones? For example, I have three drones bound to one serial usb that I can then differentiate their messages via ids. Can I use one mavsdk server to connect to all three, or do I have to run three different server instances?
```sh
mavsdk_server -p 50051 serial://com7:56000

mavsdk_server -p 50052 serial://com7:56000

mavsdk_server -p 50053 serial://com7:56000
```

JonasVautherin, 2024-04-22:
No that won't work. You should somehow find a way to demultiplex the serial stream and send the messages to their respective mavsdk_server instance.

I don't think mavlink-router can do that, but maybe you can write your own proxy that would receive MAVLink on `serial://com7:56000`, detect the different drones (I assume they have a different sysid) and forward e.g. to different UDP ports (e.g. 14550, 14551, 14552). Such that then you could run `mavsdk_server` instances on `udp://:14550`, `udp://:14551`, `udp://:14552`.

ShafSpecs, 2024-04-22:
> No that won't work. You should somehow find a way to demultiplex the serial stream and send the messages to their respective mavsdk_server instance.
>
> I don't think mavlink-router can do that, but [...]

Okay. Thanks, would read up on that 👍